pavel-demin/red-pitaya-notes #372: SPI bus not correct working with latest software.

pe0mgb, 2016-10-24:
Dear Pavel,
I have a problem with my SPI bus when I’m running the latest software. (3-10-2016) The RX Strobe signal is not available anymore. When I’m running an older version then the bus is working fine. The older version I’m using is from 27-4-2016. Is there something wrong in my setup or do I forget something to switch-on or off? The transceiver works perfect. I  added the CW connections and it works great. To work with this SDR rig is really a pleasure. Thanks again for all the work you did.
Best Regards,
Gert PE0MGB

pavel-demin, 2016-10-24:
Dear Gert,

Thank you for testing the new version and for spotting this problem.

In the new version, the I2S interface is sharing pins with the ALEX interface.  It looks like there is indeed a bug in the code for the RX strobe:
https://github.com/pavel-demin/red-pitaya-notes/blob/master/cores/axis_i2s_v1_0/axis_i2s.v#L53

Because of this bug, the RX strobe pin is mistakenly switched to input mode when the ALEX interface is activated.

To fix it, `.T(alex_flag)` should be replaced with `.T(1'b0)`. I've just fixed it with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/deff8b51ae7a85f7e141e74159d0535be5d81724).

I'll try to update the SD card image and the bazaar application later today.

Best regards,

Pavel

pavel-demin, 2016-10-24:
I've uploaded the new version to the Red Pitaya application marketplace and updated the customized SD card image available from the following link:
http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-hpsdr/

Could you please give this new version a try?

pe0mgb, 2016-10-25:
Dear Pavel,
With the latest software (24-10-2016), I have tested the Alex interface. All functions are working fine. So problem solved.
Thanks for the fast reply and solution.
Best Regards,
Gert PE0MGB

pavel-demin, 2016-10-25:
Dear Gert,

Thanks for testing this new version. I'm closing this issue.

Best regards,

Pavel

closed 2016-10-25 by pavel-demin