pavel-demin/red-pitaya-notes #374: [question] How can I transmit with 'hpsdr compatible sdr transceiver' using quisk?

df8oe, 2016-10-29:
Hi Pavel,

I am experimenting with quisk under Linux since a few days. RP works fine in RX using shipped configuration with quisk. But I am unable to TX...

"Transmit IP" was set to "0". I have tested using IP of RP but nothing changes.
And what about setting of "Transmit audio UDP"?

What I have tested: pushing PTT on GUI in AM/FM/FDV does not produce any signal output...

Here my settings in Quisk:
Hardware Type: 10
Hardware UDP port: 1024
Transmit IP: <IP of RP>
Clock frequency Hertz: 125000000
Hermes board ID: -1
IP address: <IP of RP>
Network netmask: 255.255.255.0
TX audio UDP port: 0 (???)
Hermes code version: -1

Do you have any idea to get TX working?

Best regards / 73
Andreas, DF8OE

pavel-demin, 2016-10-29:
Hi Andreas,

I've just done the following:
 - install and run Quisk
```
wget http://james.ahlstrom.name/quisk/quisk-4.1.2.tar.gz
tar -zxf quisk-4.1.2.tar.gz 
cd quisk-4.1.2
cp  hermes/quisk_conf.py ~/.quisk_conf.py 
python quisk.py 
```
 - press PTT in Quisk

TX works without any problem.

Best regards,

Pavel

df8oe, 2016-10-29:
Hi Pavel,

that works for me, too. I was gone another way: I created a new radio type "red pitaya" and choose this one. It works too but only for RX - very interesting... I will compare settings and post the reason for not working TX here.

Best regards / 73
Andreas

df8oe, 2016-10-30:
I gave up. The best way is to start using your suggestion, go to "Radio" and save this current working configuration using another name. Then this is added to the json file and can be modified during usage. I suppose one line of the "red pitaya setup configuration" is wrong - if you "add a new radio" and choose "red pitaya".

Very good work - its running perfectly!

Best regards / 73
Andreas

closed 2016-10-30 by df8oe