pavel-demin/red-pitaya-notes #375: Switching the preamp in HPSDR via a missing button

Rolf-L, 2016-10-30:
Hi Pavel,

first of all I would like to thank you for your excellent work.
At the moment I'm going to develop an interface-board for the Red Pytaya to fit it and control the PA and the LPF in my old Kenwood TRS50S TRX instead of the original TRX-boards.
What I can not find unfortunately: In the menu of HPDSR the button "Peamp on/off" seems to be missing, to switch the preamp on/off via signal DIO1 (E1 on Red Pytaya). 
My questions: 
Is it possible to insert this button in the menu of HPSDR? 
For example, there would be space next to the button "DUP" 
or how can I do switching the preamp signal?

Best regards from Schwanau in Germany
Rolf -DJ7TH-
mail(at)dj7th.de
www.dj7th.de

pavel-demin, 2016-10-30:
Hi Rolf,

The Preamp (DIO1_P) and Attenuator (DIO2_P and DIO3_P) outputs are controlled by the first ATT combo-box PowerSDR mRX PS.

Here is a table showing all the ATT settings and corresponding Preamp and Attenuator outputs:

ATT | +20 dB Preamp (DIO1_P) | -10 dB Attenuator  (DIO2_P) | -20 dB Attenuator (DIO3_P)
--- | ---------------------- | --------------------------- | --------------------------
  0 | On                     | Off                         | Off
-20 | Off                    | Off                         | Off
-10 | On                     | On                          | Off
-20 | On                     | Off                         | On 
-30 | On                     | On                          | On 
-40 | Off                    | Off                         | On 
-50 | Off                    | On                          | On 

Best regards,

Pavel

Rolf-L, 2016-10-30:
Hi Pavel,

thanks a lot for your fast replay.
I have to modify my layout only a bit, to use the settings :-)

best regards a have a nice weekend
Rolf

closed 2016-10-30 by pavel-demin

Rolf-L, 2016-12-04:
Hi Pavel,
now my preamp- and the attenuator-control works fine, but I detected that both are switched off in TX-mode and switch on again in RX-mode. Is it possible to not change the control signal (not switch off) in TX-mode? That would spare the two relays a little and the noise when switching would be reduced :-)