pavel-demin/red-pitaya-notes #376: Embedded SDR Image File not booting

zwh2, 2016-10-31:
The image file linked to from the [Embedded SDR page](http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-emb/) does not boot for me at all. I get solid blue and green LEDs and a blinking amber LED but am unable to connect over SSH or ping it.

pavel-demin, 2016-10-31:
> I get solid blue and green LEDs and a blinking amber LED

Blinking amber LED means that the FPGA configuration is loaded running.

> unable to connect over SSH or ping it

Looks like a network configuration problem.

Please try to connect to Red Pitaya using the USB/serial console as explained on the following wiki pages:
http://wiki.redpitaya.com/index.php?title=Console_connection
http://wiki.redpitaya.com/index.php?title=Windows_7_users

Then you can check the network configuration, for example using `ifconfig`.

zwh2, 2016-10-31:
Thanks for your help, I got minicom up and connected and running ifconfig shows ip address and now it is working. I don't think I changed a thing but it is now working.

Now I am trying to run the SSB example but am getting an error from grcc: 

> root@red-pitaya:~/gnuradio# grcc -d . -e trx_ssb.grc
> Unable to import Colors
> Using Volk machine: neon_hardfp_orc
> gr::log :INFO: audio source - Audio source arch: alsa
> gr::log :INFO: audio source - Audio sink arch: alsa
> Press Enter to quit: aUaUaUaUaUaUaUaUaUaUaUaUaUaUaU

pavel-demin, 2016-10-31:
I'd say that everything looks OK.

The audio underruns (aU) messages appear because the ADC clock and the USB audio clock aren't synchronized.

zwh2, 2016-10-31:
Thank you so much for your help

pavel-demin, 2016-10-31:
You're welcome.

For information. There is now a GUI for the embedded SDR. It's based on the [STM32F7 Discovery](http://www.st.com/en/evaluation-tools/32f746gdiscovery.html) board. The software for both Red Pitaya and STM32F7 Discovery can be found at [this link](https://github.com/pavel-demin/DiscoRedTRX). Some details and pictures can be found at [this link](http://www.wkiefer.de/x28/red_pitaya_stand_alone.htm). The SD card image that works with this GUI can be downloaded from [this link](https://www.dropbox.com/sh/5fy49wae6xwxa8a/AADxmf1PKHX3u8Gu0vHS2He1a/sdr/red-pitaya-gnuradio-debian-8.6-armhf-20161008.zip?dl=1).

G4DCP, 2016-10-31:
Impressed must try that I have a red pitaya and a STM32F7 Disco.

 

 

Peter.

closed 2016-12-09 by pavel-demin

Abraxas3d, 2017-01-12:
I can ping the red pitaya, and I get prompts for password, and the amber LED is blinking, but the default password for both root and redpitaya accounts (changeme) is not working.

pavel-demin, 2017-01-12:
> I can ping the red pitaya, and I get prompts for password, and the amber LED is blinking, but the default password for both root and redpitaya accounts (changeme) is not working.

What SD card image are you using?

Abraxas3d, 2017-01-12:
Trying to use yours, so that I can set up Red Pitaya with GNU Radio. 

I connected with USB cable, and was able to get in with screens. I changed the root password, and am going to try to ssh in again with the updated password. Was concerned that the default password from your instructions didn't work. More soon!

Abraxas3d, 2017-01-12:
no apt-get?

Abraxas3d, 2017-01-12:
it doesn't seem to have any of the expected utilities, so starting over.

Abraxas3d, 2017-01-12:
Much better! First image was "not well". Second time around, acting as expected.