pavel-demin/red-pitaya-notes #378: In my transceiver "PTT-in" not working properly.

pe0mgb, 2016-11-04:
Dear Pavel,
Triggered by the CW discussion I added the “PPT-in” to my transceiver. Up to now I was using for transmission the “MOX” button or the spacebar. Transceiver is working fine and I made already many many QSO’s. In USB or LSB when I make the “PPT-in” low the following happen :
-	The yellow LED #1 on RP is switched on.
-	The transceiver switched to transmission. ( e.g. antenna relays are switching and power-transistors    get their bias current )

Everything looks normal except the microphone signal. I’m missing the microphone output and so no HF output. Also on the screen I don't see the microphone spectrum.
In the CW (CWL or CWU)mode, CW together with the MOX button works fine. CW together with “PTT-in” is not working. Transceiver stays in receiving  mode.
I’m using the latest version of the software. (24-10-2016)
Do you have any idea what I’m doing wrong? I have checked the settings but I don’t see a  problem.

Thanks in advance,
Gert PE0MGB

pavel-demin, 2016-11-04:
Dear Gert,

It could be the 'Disable PTT' check-box. If it's checked, then PowerSDR mRX PS doesn't check the 'PTT-in' signal.

Here is the path to the 'Disable PTT' check-box: Setup -> General -> Options

Best regards,

Pavel

pe0mgb, 2016-11-04:
Pavel,

Thanks for the fast response but unfortunately this is not the problem. If "Disable PTT" is checked then the transceiver stays in the receiving mode. But unchecked it goes with "PTT-in" in transmission mode (USB or LSB) but no microphone output visible. I also have completely deleted and re-installed the HPSDR program again. Version 3.3.9 
I also tested the RP without my additional hardware and no  result. Using an other computer no result.
What else can I do?

Regards,
Gert PE0MGB

pavel-demin, 2016-11-04:
There is also a PTT related check-box in Setup -> Audio -> VAC 1. It's called 'Allow PTT to override/bypass VAC for Phone'. Could you please try to uncheck it?

pe0mgb, 2016-11-04:
Dear Pavel,
That’s him !!! Problem solved for USB and LSB with “PTT-in”. That was a sneaky one. Thank you very much. Problem with CW is something else and probably a similar issue. Hope you still have an advice.

Thanks again,
Gert PE0MGB

pavel-demin, 2016-11-04:
> Problem with CW is something else and probably a similar issue.

It could be yet another check-box called 'All mode Mic PTT' (Setup -> General -> Options).

pe0mgb, 2016-11-06:
Pavel,
Spend a lot of time in finding the problem. Unfortunately no luck. I feel a little bit like a monkey watching  in a rusty watch. For many of these parameters I have no clue about their meaning. Anyway, I will continue to look for the solution because I want to finish the CW portion of my transceiver. Thanks for the help.

Regards,
Gert PE0MGB

pavel-demin, 2016-11-06:
I've just checked that CW works out of the box with the default settings. Here are the steps that I've just done:
 - removed 'AppData/Roaming/FlexRadio Systems'
 - installed PowerSDR mRX PS 3.3.9.0
 - selected Hermes and Alex
 - pushed turn on button
 - pushed 40m band button
 - connected PTT In to GND
 - connected Dot In to GND

To have the max. signal amplitude, I've also set the PA gains for all the bands (6m - 160m) to 38.8 dB in Setup -> PA Settings -> PA Gain

pe0mgb, 2016-11-08:
Still bad luck. I have tried your proposal with a Win7 and Win10 machine but no good results. After your start-up proposal  when I connect GND to PPT-in and CW, the RP stay even in receiving mode. Nothing happen. Of course their LEDS are on.

pe0mgb, 2016-11-13:
Pavel,
May I have a copy of your HPSDR program database? Just to be sure that my program is running exact the same way yours is doing.
Regards,
Gert PE0MGB

pavel-demin, 2016-11-13:
Here is a copy of my configuration directory:
https://www.dropbox.com/sh/5fy49wae6xwxa8a/AAB0fItqBJRKskdlRdeR5xZva/FlexRadio%20Systems.zip?dl=1

Before compressing it, I've checked that it works with PTT In and Dot In in CW mode.

pavel-demin, 2016-11-13:
> After your start-up proposal when I connect GND to PPT-in and CW, the RP stays even in receiving mode.

It's just occurred to me that maybe we aren't looking at the same problem. What do you mean by "RP stays in receiving mode"? What is your definition of the the "receiving mode" or "transmitting mode"?

Here is how I do my tests:
 - set CWL mode in the program
 - connect PTT In to GND
 - connect Dot In to GND
 - check if there is a signal coming from OUT1

It just works with the default settings.

pavel-demin, 2016-11-13:
Just noticed that in the CW modes PowerSDR mRX PS doesn't activate PTT Out when PTT In is active.

The "Firmware Keyer" button that controls this behavior is invisible. It's possible to uncheck it via database.xml:
```
  <State>
    <Key>chkCWFWKeyer</Key>
    <Value>False</Value>
  </State> 
```

pe0mgb, 2016-11-14:
Hi Pavel,
Thanks for all the effort. I have changed the my database according your proposal.
Now it works. PTT out is low after PTT in went low. Unfortunately there is no output anymore when the Dot is low. Also using the “mox” button together with DOT to low gives no output.
When I changed the database back to the previous state then I have output again with MOX and DOT to low.

pe0mgb, 2016-11-21:
Hello Pavel,
Maybe you are still waiting for an answer. 
With "RP stays in receiving mode" I mean that the receiver is working and  I get a signal from the speaker.
With "transmitting mode"? I mean that PTT out is going low and HF appears at connector out1.

After :
  <State>
    <Key>chkCWFWKeyer</Key>
    <Value>False</Value>
  </State> 

•	set CWL mode in the program
•	connect PTT In to GND
•	connect Dot In to GND

No output at OUT1 but PTT out is now “low”. Before the change in the database with same commands PTT out stays “high”  and also no output at OUT1

pavel-demin, 2016-11-21:
Hello Gert,

Thanks for the answers.

I agree that changing chkCWFWKeyer isn't a solution.

When I do my tests of the CW mode, I see a CW signal at OUT1, but PTT out doesn't change its state.

The only solution for the PTT Out problem that I can think of is to connect PTT In to PTT Out in my code running on Red Pitaya. It already does something similar for Preamp Out:

https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_hpsdr/server/sdr-transceiver-hpsdr.c#L623

For PTT Out, it can be done by replacing
```
*gpio_out = (frame[2] & 0x1e) << 3 | att << 2 | preamp << 1 | ptt;
```
with
```
*gpio_out = (frame[2] & 0x1e) << 3 | att << 2 | preamp << 1 | (ptt | (*gpio_in & 1));
```

 

Best regards,

Pavel

pe0mgb, 2016-11-21:
For other modes like LSB and USB, PTT out and PTT in are working perfect.

pavel-demin, 2016-12-08:
In the latest version, PTT In is directly connected to PTT Out. The CW keyer in this version can work without pressing MOX or PTT In. Could you give this new version a try?

ka6s, 2016-12-09:
Pavel,

I've checked it out - seems to work VERY nice!

No more Foot Switch - which makes it a real joy to use!

Steve

pe0mgb, 2016-12-09:
Dear Pavel,
It all works now very well. No problems anymore with PTT in etc. It’s a pleasure to work with it now.
Is there still a possibility to use a regular CW key?
Thanks for all the work.
Best Regards,
Gert PE0MGB

pavel-demin, 2016-12-09:
> It all works now very well.

Great! Thanks for checking.

> Is there still a possibility to use a regular CW key?

Yes. The 'iambic' checkbox should be unchecked.

closed 2016-12-09 by pavel-demin