pavel-demin/red-pitaya-notes #380: Gracefull shutdown by means of a push button

on3vna, 2016-11-07:
GM Pavel,

Is it possible to have a "gracefull shutdown" option by the means of a switch?
Now you have to do a shutdown via ssh of serial interface before you unplug the
power.
I had already 2 times "Murphy" hitting my disk with total loss when cutting the power
before doing the proper shutdown.

Thanks
Johan
73 de ON3VNA

pavel-demin, 2016-11-07:
A couple of interesting link on the subject:

https://github.com/andreamerello/linux-zynq-stable/issues/12

http://lists.infradead.org/pipermail/linux-arm-kernel/2015-May/341037.html

pavel-demin, 2016-11-07:
The main problem is where to connect the button.

on3vna, 2016-11-07:
Sacrifice one of the hermes ctrl pins (4)  maybe ?

pavel-demin, 2016-11-07:
Another possibility could be to use the Dash and Dot inputs. For example, connecting Dash and Dot to GND for more than 3 seconds would shutdown the system.

The software part could be done as described in the following article:

https://www.element14.com/community/docs/DOC-78055/l/adding-a-shutdown-button-to-the-raspberry-pi-b

on3vna, 2016-11-07:
I would leave dash and dot as is.

Are the hermes crtl lines not available via SPI and I2C interface also ?

ka6s, 2016-11-07:
I'll vote to keep the Hermes CTL pins. They eliminated the need for I2C
decode to do this function.  They are the fundamental controls required to
control just about any external filters I've found. You have to change
their voltage level to control some external filters -but pretty much all
of them just take the 4 wires.

I'll also point out that you can really only control the power-down - how
do you power up?

Steve

on3vna, 2016-11-07:
GM Steve,

It seems we running out of pins -;).
Powerup should be just putting in the power and it should start, i still have to try once with the HPSDR tranceiver image to see if that works, when using the ecosystem it's really a cumbersome action, connecting with web brouwer starting up HPSDR on the RED PI  keep that window open ...
I need to do some more testing on it.

Rgds
Johan

pavel-demin, 2016-11-07:
> i still have to try once with the HPSDR tranceiver image to see if that works, when using the ecosystem it's really a cumbersome action, connecting with web brouwer starting up HPSDR on the RED PI keep that window open ...

Actually, the small images (based on Buildroot) for the SDR transceivers don't need the shutdown button. Their file system is mounted read-only and it's safe to cut power at any moment.

Maybe a solution for the large images (based on Debian) would be to use RAM disks for /tmp and /var and mount the partitions from the SD card as read-only.

If I'm not mistaken, it could be done by modifying `/etc/fstab` in the following way:

```
/dev/mmcblk0p2  /               ext4    defaults,ro   0       1
/dev/mmcblk0p1  /boot           vfat    defaults,ro   0       2
tmpfs           /tmp            tmpfs   defaults      0       0
tmpfs           /var            tmpfs   defaults      0       0
```

More information can be found at the following links:

https://wiki.debian.org/ReadonlyRoot
http://www.zdnet.com/article/raspberry-pi-extending-the-life-of-the-sd-card

ka6s, 2016-11-07:
> life-of-the-sd-card

I use the debian development version - and reboot my box 50 times a day -
using ssh.  Haven't lost it yet Once in awhlie I use the power switch - but
usually just login and reboot.

An SRAM version of things would be less convenient - at least when I am
trying to fix something - but you could do worse.

Steve

on3vna, 2016-11-08:
Steve,

rebooting via SSH is not the issue, it is just that moment in time when you just pull the powerplug out and our dear friend Murphy is standing around the corner...

An SRAM version indeed is not convenient when you are in the debugging or testing stage i agree with that. I can imagine me already testing in that case .... I think the neighbors would hear it...

It is just a request for improvement.

Johan

pavel-demin, 2016-12-17:
Just found a very interesting solution for a read-only file system based on unionfs:
http://blog.pi3g.com/2014/04/make-raspbian-system-read-only
http://unionfs.filesystems.org

jeanminet, 2016-12-19:
I Pavel, I tried the unionfs approach and it works like a charm!
https://github.com/Koheron/koheron-sdk/pull/367/files
Best,
Jean

ted051, 2017-08-21:
Hi Pavel,
for this problem I suggest the following solution:
If Dot In (DIO2_N) and Dash In (DIO1_N) are both connected to ground **more than 10 seconds**, the system is going to shutdown.
This would allow users to work without an additional computer.
Best regards, Wolfgang

pavel-demin, 2017-09-23:
With the new SD card image based on [Alpine Linux](https://alpinelinux.org), the SD card is mounted read-only and nothing is written to the SD card during its normal operation. So, the power can be switched off at any moment without damaging the file system on the SD card.

This new SD card image will replace (or already replaces) all my customized SD card images based on the 0.95 ecosystem and all the SD card images based on Debian.

Since this new SD card image doesn't require graceful shutdown, I'm closing this issue.

closed 2017-09-23 by pavel-demin