pavel-demin/red-pitaya-notes #381: HPSDR Thetis Protocol

ptamike, 2016-11-13:
Hi Pavel,

I'm enjoying using the Red Pitaya with your software but it could be even better if it supported the HPSDR Thetis protocol. Simon Brown is supporting this with his excellent SDR-Console V3 software and will be including full transceive operation.

Thanks for your hard work on this project,

Mike - G4WNC

pavel-demin, 2016-11-13:
Hi Mike,

Thank you for your encouraging feedback for your kind words. I'm glad that you're enjoying my software.

Currently, I'm not very interested in implementing the new HPSDR protocol. There is still no TX capable software that supports it: SDR-Console V3 is still a preview with disabled TX, Thetis isn't released yet. 

Best regards,

Pavel

pavel-demin, 2017-03-03:
Just found that Thetis installers can be downloaded from the following link:
http://svn.tapr.org/repos_sdr_hpsdr/trunk/W5WC/Thetis_Installers/

ka6s, 2017-03-03:
Pavel - My understanding is that pihpsdr does support transmit for the new
protocol.

Steve

http://svn.tapr.org/repos_sdr_hpsdr/trunk/W5WC/Thetis_Installers/

closed 2018-01-23 by pavel-demin