pavel-demin/red-pitaya-notes #385: Openhpsdr android app not working anymore

pa3hea, 2016-11-28:
Hello, 
After the last software update of the redpitaya, the android version of openhpsdr cannot start a session anymore.
It sees the the redpitaya but if you try to start a session there is no waterfall and non of the button's are functional.
When you close the session you need to restart the webinterface to reconnect.
I am not a programmer so I don't no where it go's wrong but it is a pitty that I can't use my tablet anymore.
Is there a solution for this problem, I can't find anything about it on the internet or forums.

pavel-demin, 2016-11-28:
Unfortunately, I don't have any device with the Android OS capable to run the OpenHPSDR application. So, I can't reproduce this problem.

As a first step, we should try to find what was the last version that worked with the OpenHPSDR application.

pa3hea, 2016-11-28:
Hello Pavel, thanks for your fast reply.
The previous version was working fine.
It has something to do with the webinterface I think, without that you don't see any redpitaya available.
Maybe you can compare the differences in both versions.

Tnx 

Bart

pavel-demin, 2016-11-28:
> The previous version was working fine.

Do you have by chance the exact number of the previous version that worked?

The current version is 1354. The only modification in this version is related to the I2S audio codec: https://github.com/pavel-demin/red-pitaya-notes/commit/e6bcfe06d8e7f9191cce2b8f7463f82f81b0d3b0

The last time I did some significant changes in the protocol implementation was between versions 1191 and 1303: https://github.com/pavel-demin/red-pitaya-notes/commit/6702ab854765afa866e93e1a01a961226bb1e87d

Both versions can be downloaded from the following links:
https://www.dropbox.com/sh/5fy49wae6xwxa8a/AAD_dwEXtUsXmqD-gaAafUVQa/apps/sdr_transceiver_hpsdr-0.94-1191.zip?dl=1
https://www.dropbox.com/sh/5fy49wae6xwxa8a/AADXR2_t1Wt3dTeDSFloYrLba/apps/sdr_transceiver_hpsdr-0.94-1303.zip?dl=1

Could you please check how these two versions work with the OpenHPSDR application?

pa3hea, 2016-11-28:
Hello Pavel,

Thanks for trying to solve this issue I appreciate it very much.
Version 1191 is working.
Version 1303 is not working.
Somewhere in the 1303 version went something wrong.

pavel-demin, 2016-11-28:
Thanks for the test. It's good that you've identified the last version that worked with the OpenHPSDR application.

I'll try to do some tests with the desktop version of the OpenHPSDR application:
http://g0orx.blogspot.com/2015/04/java-desktop-application-based-on.html

pa3hea, 2016-11-29:
Hello Pavel,

Sorry for the late reply, I am a bussy guy.
I never used the java application before so I don't no if the same problem exist with that.
I downloaded it to look if it runs in Linux Mint 18, it does, but I can't start a session at all.
Version 1191 and 1303 both not working with the java application.
I don't no if this is a problem in my laptop or in the webinterface, so I can't help you with information about this issue.

pa3hea, 2016-11-29:
Hello Pavel,

I have some information for you.
There are probably some other issues in my linux installation when I try to start a session with the java desktop application, so if you ask me just focus to windows at this moment.
I try to solves the Linux problem on a later time.

I installed svn for windows and I download de java desktop application of openhpsdr.
I can tell you that it is possible to start a session with version 1191 and not with version 1303.
So I guess the same problem with the openhpsdr android app exists with the java desktop application to.
F.Y.I. my windows version is windows 10 pro 64bit nl.
I hope this is useful for you.

pavel-demin, 2016-11-30:
Thanks for the tests.

I've also reproduced the problem with the OpenHPSDR application running under Ubuntu 14.04.

The problem was caused by the recvmmsg() timeout misbehavior described at [this link](https://bugzilla.kernel.org/show_bug.cgi?id=75371).

I've added the suggested workaround with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/9420c3860400739fb83a39e0a7bef67db51bc53d).

pa3hea, 2016-11-30:
Hi there Pavel,

Great to hear that in such a short time you find a solution for this issue, which is very much appreciated.
Is this fix added to the version downloadable in the redpitaya store, or do I need to add the workaround manually.
Anyhow I will thank you in advance for your effort that you putt in this project.
It encouraged me to try to build a SDR transceiver.

tnx

Bart

pavel-demin, 2016-11-30:
> Is this fix added to the version downloadable in the redpitaya store

Not yet. I'll try to update the pre-built binaries later today.

pavel-demin, 2016-11-30:
I uploaded the new version to the Red Pitaya application marketplace and updated the customized SD image. Could you please check if it works with the OpenHPSDR application for Android?

pa3hea, 2016-11-30:
Hi Pavel,

I upgrade the Red Pitaya software to version 1398 and tested the Openhpsdr Android app and the Openhpsdr java desktop app running on Windows 10 pro 64bit NL, and I can tell you that both pieces of software are working like it should be.
You did a great job Pavel, I'm playing around for some time with it and so far I can see it's stable to.
So I think case closed for now.
Well done...

tnx

Bart

pavel-demin, 2016-11-30:
Great! Thanks for all the tests!

closed 2016-11-30 by pavel-demin