pavel-demin/red-pitaya-notes #401: SDR RX HPSDR + CWSL + HermesIntf + Skimsrvr

azwirko, 2016-12-08:
HI - I realize my problem involves several pieces of software and not sure which may be at fault if any, but thought I'd post and ask if anyone else has experienced my issue.  When running the SDR receiver HPSDR and using HermesIntf 16.6.27 https://sourceforge.net/projects/hermesintf/files/ and CWSL_Tee https://github.com/HrochL/CWSL with CW Skimmer Server 1.5 http://www.dxatlas.com/skimserver/ , I occasionally see a CW call decode repeated on a wrong band.  This only happens when I activate 6 receivers in CW Skimmer Server.  If I run 5 or less I don't experience the problem.  Specifically, a station received on the second to highest band I'm monitoring is occasionally spotted on the highest band enabled.  For example in the attachment I'm monitoring 1.8, 3.5, 7.0, 10.1, 14.0 and 18.0 MHz.  Some stations I know are transmitting on 14.0 MHz are spotted there AND on 18.0 MHz.  It doesn't appear to be a RF overload issue as some stations are weak (5 dB) while others louder (24 dB).  This does not happen for all stations though.  See attachment as some 14 MHz are spotted properly on only that one band.  This seems regardless if the station is labeled with CQ, DE or none.  Note that my other tests track this same behavior  regardless of bands selected for monitoring in CW Skimmer server.  For example if I monitor 7, 10, 14, 18, 21 and 24 MHz I will occasionally see a spotted station on 21 MHz also spotted on 24 MHz.  The offset from the bottom of each CW band edge tracks with the original station's band.  

One other note I also run PowerSDR mRX with Redpitaya and your HPSDR receiver and in the PSDR hardware Setup I see that the Redpitaya reports as a v2.5 Hermes.  The writer of HermesIntf README on Sourceforge states true Hermes v2.4&2.5 boards run only 5 receivers.  He also states your receiver runs and is supported with 6 receivers.  So, I'm not sure if this is a documentation issue, or some software issue.

Thanks for all your work on the Redpitaya - you've got some excellent software! I've enjoyed trying all your projects.

73 

andyz - K1RA
http://k1ra.us/

p.s. - I've CC'd this note to the writer of HermesIntf 
[image]
[image]
[image]

pavel-demin, 2016-12-08:
Hi Andy,

Thank you for your tests and for spotting this problem.

One of the earliest versions of the 6-band receiver had a similar with all the spots duplicated on the last two bands. The problem was discussed in https://github.com/pavel-demin/red-pitaya-notes/issues/265.

What is strange with this new problem is that not all the spots on the last two bands are duplicated.

HermesIntf doesn't use the Hermes version to  identify Red Pitaya. It uses a special ID tag that was added in https://github.com/pavel-demin/red-pitaya-notes/commit/aaa44e0b6377aa7b53e51381e0d69dfe56e23ac1.

Since the only difference with the previously tested configuration is CWSL, I'd suggest to redo the test with and without CWSL.

Best regards,

Pavel

k3it, 2016-12-09:
I agree  this does not look like duplication of receivers as in #265 . When IQ streams are identical on different bands my guess is that the number of CW decoders should be similar, but in this case you have 30 decoders on 20m and only 6 on 17m.   I looked through hermesintf code and nothing jumped out.  

 If removing CWSL layer does not help,  I hope Alex @VE3NEA had seen something like this before and can help.

pavel-demin, 2016-12-09:
I've just reread more carefully the first post and discovered the following part:

> I have removed CWSL_Tee from the configuration and noted the same duplicate problem, so that rules out that piece of software.

So, the problem isn't in CWSL.

I've also quickly looked through my code but could not find anything suspicious.

pavel-demin, 2016-12-09:
What would be also interesting to understand is whether the duplicated spots appear immediately after some changes in the CW Skimmer Server configuration or not.

I could imagine that because of some buffering I/Q streams could be flowing into wrong bands for a shot time just after the configuration changes.

azwirko, 2016-12-09:
Pavel & K3IT - thank you for taking time to look at this, but I'm sorry to say it looks like I was running an older image that may have had the original #265 bug.  My Redpitaya is remote and I can't reburn the SDcard, but I see file stamps on the bit and server files as follows:

/opt
-rwxr-xr-x    1 root     root       1119056 May  3  2016 sdr_receiver_hpsdr.bit
/opt/bin
-rwxr-xr-x    1 root     root        689464 May  3  2016 sdr-receiver-hpsdr

I took a chance and I pulled the latest .ZIP file from git and SCP'd and overwrote the old files and rebooted the system with these:

/opt
-rwxr-xr-x    1 root     root       1198264 Jun 28  2016 sdr_receiver_hpsdr.bit
/opt/bin
-rwxr-xr-x    1 root     root        692952 Jun 28  2016 sdr-receiver-hpsdr

Problem solved! I'm running the complete chain of HPSDR, CWSL, Hermes and CW Skimmer Server as stated earlier with no more duplicates.  Thanks guys for your time and some great pieces of s/w.

73

andyz - K1RA

closed 2016-12-09 by azwirko