pavel-demin/red-pitaya-notes #404: MCPHA - negative pulse

huml, 2016-12-12:
Hi Pavel,
I am really grateful for your MCPHA. It works like a charm. Do you think it is possible to add an option to process negative pulses? It is useful for cases, when you want to process pulses directly  from a detector not using inverting amplifier.

Yours sincerfely
Ondrej Huml

pavel-demin, 2016-12-12:
Hi Ondrej,

Thank you for you kind feedback. I'm glad that my MCPHA application works well for you.

I'll add an option to process negative pulses. I'll let you know when it's ready.

Best regards,

Pavel

huml, 2016-12-12:
Thank you for such a fast response.

If you are interested, I am sending you a datafile with 3.5 hour
measurement of Cs-137 using scintilation detector.

Best regards
Ondrej

pavel-demin, 2016-12-12:
There is now a new version that works with negative pulses. Both the application running on Red Pitaya and the client program should be updated.

closed 2016-12-21 by pavel-demin