pavel-demin/red-pitaya-notes #416: HDSDR on Win10: ExtIO, Load DLL failed

hb9pae, 2016-12-19:
Hi Pavel,
I'm enjoying using the Red Pitaya with your excelent software, thank you for your effort!
After new intstallation of my computer, HDSDR doaes not load the DLL "ExtIO_RedPitaya_TRX.DLL" anymore.  (ExtIO, Load DLL failed), same behavior with SDR#.

I belive, that an older Version of the DLL worked well.
Source of DLL: Red Pitaya notes > sdr-transceiver > 
https://www.dropbox.com/sh/5fy49wae6xwxa8a/AAABy8xWrv4wFFbkYfmVU6DGa/sdr/ExtIO_RedPitaya_TRX.dll?dl=1

MD5Hash of used ExtIO File: a96f9f84ddc6beb49796b509e7ea1f87

Thanks for any suggestion.

73 de Peter, HB9PAE

pavel-demin, 2016-12-19:
Hi Peter,

The current version of ExtIO_RedPitaya_TRX.dll was built in November 2015. Your MD5 sum is correct.

Could you check if Microsoft Visual C++ 2008 SP1 Redistributable Package is installed on your system? If it's not installed, could you try to download it from the following link and install it? 
https://www.microsoft.com/en-us/download/details.aspx?id=5582

Best regards,

Pavel

hb9pae, 2016-12-19:
Microsoft Visual C++ 2008 SP1 Redistributable Package is installed, but same behavior. SDR# does not load the DLL as well.

Windows 10 (32 bit)
- I tried HDSDR V 2.70 and the beta V 2.75, same negative result. (MSG: ExtIO, Load DLL failed)
- Windows 10 Systemcheck SFC runs successful

On Windows 7, HDSDR loads DLL well.

pavel-demin, 2016-12-19:
The following two discussions mention a strange but working solution for the  Microsoft Visual C++ 2008 and Windows 10 compatibility problem:
https://answers.microsoft.com/en-us/windows/forum/windows_10-other_settings/error-error-997-while-installing-microsoft-visual/60a4f08e-a740-475d-8858-3435b2784b01
https://answers.microsoft.com/en-us/windows/forum/windows_10-other_settings/windows-10-will-not-install-visual-c-20082010/f7a5dadd-0ca0-4393-87c5-c44e9e305408

If I'm not mistaken, the idea is to the folder C:\ProgramData\Microsoft\Crypto\RSA\S-1-5-18 to C:\ProgramData\Microsoft\Crypto\RSA\S-1-5-18_old and then to reinstall Microsoft Visual C++ 2008 and/or its redistributable package.

hb9pae, 2016-12-20:
Hi Pavel,
Thanks for the MS Support-Links.
As I understand, the above Support-links covers about install problems with  Microsoft Visual C++ 2008  Package. On my W10 System, I can install those packages without any problems.

Nevertheless I followed the procedure:
-   renamed the folder  C:\ProgramData\Microsoft\Crypto\RSA\S-1-5-18
-  deinstalled the old Microsoft Visual C++ 2008  Package
-  reinstalled Microsoft Visual C++ 2008  Package newly

Starting HDSDR:  ExtIO, Load DLL failed

I will do test with "Clean Windows Start" as mentioned in the Support-Note

73 de Peter, hb9pae

pavel-demin, 2016-12-20:
I've rebuilt this ExtIO DLL using Visual Studio 2015:
https://www.dropbox.com/sh/5fy49wae6xwxa8a/AAB3hpIljgT_3CEKa0ZBCrhba/sdr/ExtIO_RedPitaya_TRX_2015.dll?dl=1

hb9pae, 2016-12-20:
Hi Pavel,

Using the new DLL (2015) HDSDR works great also on Win10!

Thanks so much for your support!

73 de Peter HB9PAE

closed 2016-12-20 by hb9pae

pavel-demin, 2016-12-21:
Hi Peter,

Thanks for the test. I've replaced the old ExtIO_RedPitaya_TRX.dll with the new DLL (2015).

Best regards,

Pavel