pavel-demin/red-pitaya-notes #426: awk: symbol lookup error: awk: undefined symbol: mpfr_z_sub

dragniko, 2016-12-27:
vivado -nolog -nojournal -mode batch -source scripts/core.tcl -tclargs axi_axis_reader_v1_0 xc7z010clg400-1

awk: symbol lookup error: awk: undefined symbol: mpfr_z_sub

In Makefile emptying LD_LIBRARY_PATH variable worked for me with Vivado 2016.2, now I am using 2016.4 and this issue with awk is resurrected ...

# solves problem with awk while building linux kernel
# solution taken from http://www.googoolia.com/wp/2015/04/21/awk-symbol-lookup-error-awk-undefined-symbol-mpfr_z_sub/
LD_LIBRARY_PATH =

Any idea how to fix this error?

pavel-demin, 2016-12-27:
What OS are you using? I can't reproduce this problem with Debian 8.6.

dragniko, 2016-12-27:
cat /etc/debian_version
8.6
uname -a
Linux debian 3.16.0-4-amd64 #1 SMP Debian 3.16.36-1+deb8u2 (2016-10-19) 
x86_64 GNU/Linux

I followed your instructions at
http://pavel-demin.github.io/red-pitaya-notes/led-blinker/

except I installed one additional package: gcc-arm-none-eabi when 
creating debian development vbox;

Cheers
Dragan

pavel-demin, 2016-12-27:
I've just checked that Vivado 2016.4 doesn't set LD_LIBRARY_PATH. So, I'd say that the solution would be to switch to Vivado 2016.4.

dragniko, 2016-12-27:
I need to be able to reproduce old successful build using Vivado 2016.2 
first;
how can I clone your red-pitaya-notes git repository that has only 
2016.2 patches,
prior to 2016.3 and 2016.4 changes?
Once I get this old build to work, I can gradually go to 2016.4 and see 
if things work again;
Thanks
Dragan

pavel-demin, 2016-12-27:
> how can I clone your red-pitaya-notes git repository that has only 2016.2 patches

```
git clone https://github.com/pavel-demin/red-pitaya-notes.git
cd red-pitaya-notes
git checkout 17a63864b27b6903e2ac10552097ddb702a34507
```

dragniko, 2016-12-27:
This was very helpful, thank a lot; this way I do not need to make my 
local copies
of distributions that worked; can I get checkout signatures for 2016.3 
and 2016.4 branches as well?
In this way I can automatically rollback to previous version I know my 
build works;
so far build 2016.2 with your checkout works flawlessly;
Cheers
Dragan

dragniko, 2016-12-27:
Hi Pavel,
can you please add

+CONFIG_USB_ACM=y

to 2016.4 patch so that /dev/ttyACM0 device is created?
I do it using sed on your
linux-xlnx-xilinx-v2016.4.patch
but it will be nice if you can already include by yourself.
Thanks a lot
Dragan

pavel-demin, 2016-12-27:
> can I get checkout signatures for 2016.3 and 2016.4 branches as well?

I've just created two branches 2016.2 and 2016.3. So, switching to the 2016.2 compatible version can be done with `git checkout 2016.2`. The master branch corresponds to 2016.4.

> can you please add +CONFIG_USB_ACM=y 

I've added it with https://github.com/pavel-demin/red-pitaya-notes/commit/3f74cab470154d6834be22a2e45b3445ce0f784f.

closed 2017-01-03 by pavel-demin

jeanminet, 2017-01-03:
I had the same `awk` problem with Vivado 2016.4 on Ubuntu 2016.4.

Using the Ubuntu version of `libmpfr.so.4` instead of the one provided by Xilinx seems to solve the problem:

```bash
path=/opt/Xilinx/Vivado/2016.4/lib/lnx64.o
mv $path/libmpfr.so.4 $path/libmpfr.so.4.orig
ln -s /usr/lib/x86_64-linux-gnu/libmpfr.so.4 $path/libmpfr.so.4
```

http://perso.utinam.cnrs.fr/~emeyer/doku.php?id=wiki:linux_pour_zynq_de_xilinx:installation_petalinux#utilisattion

pavel-demin, 2017-01-03:
Thanks for the hint. However, I don't understand why I can't reproduce this problem. In my installation of Vivado 2016.4, the LD_LIBRARY_PATH variable isn't set:
```
# source /opt/Xilinx/Vivado/2016.4/settings64.sh
# echo $LD_LIBRARY_PATH

```

In Vivado 2016.2, LD_LIBRARY_PATH was set in /opt/Xilinx/Vivado/2016.2/.settings64-Vivado.sh

In Vivado 2016.4, the lines setting LD_LIBRARY_PATH were removed from /opt/Xilinx/Vivado/2016.4/.settings64-Vivado.sh.

How awk finds libmpfr.so.4 in /opt/Xilinx/Vivado/2016.4/lib/lnx64.o?

jeanminet, 2017-01-03:
I have no idea :(

dragniko, 2017-01-03:
Yup!
this does solve the problem, thanks a lot Jean!
Cheers
Dragan

klyone, 2017-03-08:
Hi @jeanminet . Have you had any problem to lauch sdk? I have applied the solution you discussed but my SDK does not open... and the error messages is the same "awk: symbol lookup error: awk: undefined symbol: mpfr_z_sub".

Any clue about it?

Thanks in advance.

jeanminet, 2017-03-08:
Hi @klyone 
The solution I described works on my installation (Ubuntu 16.04) but it does not seem to very reproducible across different installations.
Anyway, the error message `awk: symbol lookup error: awk: undefined symbol: mpfr_z_sub` does not seem to affect the build.

klyone, 2017-03-08:
Hi @jeanminet Thanks for your answer. My problem is that SDK does not boot properly. I found this solution to fix it [1]. The "awk message" is appearing still but now I can work with SDK. I hope this information is useful for you.

[1]: https://forums.xilinx.com/t5/Embedded-Development-Tools/Vivado-2016-3-SDK-launches-with-a-segmentation-fault/td-p/728638