pavel-demin/red-pitaya-notes #444: Make SD card format clear in docs

laci37, 2017-01-07:
The original Red Pitaya SD card comes with two partitions on it. I wanted to try out the SDR transceiver. I overwrote the SD cards larger partition with the contents of the zip file on the page, now my Pitaya does not seem to boot. What is the correct format for the SD cards partitions? I think that the docs should be clearer on what is supposed to be on the SD card. Maybe provide binary images?

pavel-demin, 2017-01-07:
> The original Red Pitaya SD card comes with two partitions on it. I wanted to try out the SDR transceiver.

It's possible to install the SDR transceiver application to the original Red Pitaya SD card. It can be done from the web interface via the application marketplace.

>  I overwrote the SD cards larger partition with the contents of the zip file on the page...

I'd suggest to unpack the .zip file to the first (smaller) partition.

> What is the correct format for the SD cards partitions?

Empty SD card with a single FAT formatted partition. Just as almost any brand new card.

> Maybe provide binary images?

Tried in the very beginning. Users also complained and asked for a simple .zip file. So, this .zip file is for those who prefer to not install the SD card image provided by Red Pitaya.

pavel-demin, 2017-01-07:
> I think that the docs should be clearer on what is supposed to be on the SD card.

Could you propose a phrase what would be clearer and what I could add to my notes?

closed 2017-01-10 by pavel-demin