pavel-demin/red-pitaya-notes #449: Modprobe with kernel 4.6.0-xilinx

hmaarrfk, 2017-01-15:
Hey Pavel,

I'm trying to get a USB Wifi dongle working.

It sems that the linux drivers are included in your image, but modprobe seems to be problematic.

It is giving me the following error codes:

```sh
root@red-pitaya:~# lsusb
Bus 001 Device 002: ID 0a5c:bd1e Broadcom Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```sh
root@red-pitaya:~# modprobe b43
modprobe: ERROR: ../libkmod/libkmod.c:557 kmod_search_moddep() could not open moddep file '/lib/modules/4.6.0-xilinx/modules.dep.bin'
```

Your help is appreciated!

Mark

pavel-demin, 2017-01-15:
Hi Mark,

I link all the modules statically to the kernel and I don't create the directories required for modprobe.

If you need loadable modules, then '=y' should be replaced with '=m' in my patch file:
https://github.com/pavel-demin/red-pitaya-notes/blob/master/patches/linux-xlnx-xilinx-v2016.4.patch

Some instructions on where and how to copy the loadable modules can be found at the following link:
https://github.com/pavel-demin/red-pitaya-notes/tree/master/projects/bare_metal_test

Best regards,

Pavel

pavel-demin, 2017-01-15:
Looks like the driver that supports `0a5c:bd1e` is brcm80211 and it's not enabled in my patches.

pavel-demin, 2017-01-15:
I've just enabled brcm80211 with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/4e4871f390f731caf32cf9399ceac1edc55c194a).

Here is a link to the updated uImage file:
https://www.dropbox.com/sh/5fy49wae6xwxa8a/AADkv4FD7XB4HuFP5ywW3iQka/uImage?dl=1

hmaarrfk, 2017-01-16:
Hi Pavel,

Thank you for the quick replies. 
I'm a little new to embedded linux. Is there any reason not to enable loadable modules?

I replaced the `uImage` file in my /boot partition with the one you provided. A few bugs:
1. It seems to be stuck starting the `isc-dhcp-server.service` if it is plugged in. I was able to boot it up if I leave the USB unplugged.
2. Once I plug it in, it gives the following error message
```sh
usb 1-1: new high-speed USB device number 2 using ci_hdrc
usb 1-1: Direct firmware load for brcm/brcmfmac43143.bin failed with error -2
```

Thanks again,

Mark

hmaarrfk, 2017-01-16:
Ok, I seem to have gotten the interface showing up on ifconfig.
It seems that debian jessie doesn't include the usb drivers in `firmware-brcm80211`.
You have to download them manually. The instructions [here](https://www.raspberrypi.org/blog/new-raspbian-and-noobs-releases/#comment-1040775) (RaspberryPi Forum) helped.

I would simplify them to:
```sh
git clone --depth 1 git://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git
cp -R linux-firmware/brcm /lib/firmware/.
```
should be enough. I only moved the `brcm/brcmfmac43143.bin` file in place.

I also disabled the dhcp server so that the zynq would actually boot without hanging for ever with the USB plugged in.
```sh
systemctl disable isc-dhcp-server.service
```

hmaarrfk, 2017-01-16:
FYI, it also seems imperative to have the correct timezone set for sshd to work correctly.

One can reconfigure the timezone by issuing a `dpkg-reconfigure tzdata` command.

pavel-demin, 2017-01-16:
> Is there any reason not to enable loadable modules?

The loadable modules are enabled. I just don't add them to my SD card images.

At the moment, I find it easier to work with the statically linked modules.

For example, if I have a SD card with the root file system in a ramdisk image, then I can add new drivers just by replacing uImage without unpacking/repacking the ramdisk image.

pavel-demin, 2017-01-16:
> I also disabled the dhcp server so that the zynq would actually boot without hanging for ever with the USB plugged in.

Without this service the access point won't work properly. I think that once you have working firmware for the USB WiFi adapter, the dhcp server can be re-enabled.

hmaarrfk, 2017-01-17:
Ahhhh, do you have the RedPitaya behave as an access point for both Ethernet and Wifi?

My router was set to 192.168.100.X and I thought there was some weird bug with automatic IP addresses being assigned. Maybe you have the default IPs set to 192.168.100.X as well on the RedPitaya. I'll have a look at the settings tonight.

pavel-demin, 2017-01-17:
> do you have the RedPitaya behave as an access point for both Ethernet and Wifi?

Only for WiFi.

> Maybe you have the default IPs set to 192.168.100.X as well on the RedPitaya.

WiFi network is 192.168.42.X:

https://github.com/pavel-demin/red-pitaya-notes/blob/master/scripts/debian.sh#L189

hmaarrfk, 2017-01-17:
Hmm thanks,

The `eth0` interface seems to be set to [static IP](https://github.com/pavel-demin/red-pitaya-notes/blob/master/scripts/debian.sh#L204) at 192.168.1.100 (got that order wrong). I understand the choice. A possible improvement would be to make that explicit in the LED blinker application (or where appropriate).

Honestly though, great work, your tutorials are quite easy to follow and help build confidence.

pavel-demin, 2017-01-17:
> The eth0 interface seems to be set to static IP at 192.168.1.100.

DHCP client uses this static address if it doesn't receive any response from a DHCP server. I added this option to emulate the original Red Pitaya behavior:
http://wiki.redpitaya.com/index.php?title=OLD_OS_User_Manual#Network_DHCP_configuration

pavel-demin, 2017-09-23:
The firmware files that were missing from the SD card image based on Debian are now included with the new SD card image based on Alpine:
http://pavel-demin.github.io/red-pitaya-notes/alpine

This new SD card image will replace all my SD card images based on Debian.

closed 2017-09-23 by pavel-demin