pavel-demin/red-pitaya-notes #451: How hook up MIKROE-506 board

n1gp, 2017-01-18:
Hi Pavel,

I bought a MIKROE-506 board and am trying to get audio in/out.

I tried hooking the 3.3/gnd and I2C lines up but no audio heard. Then

hooked up SPI but still nothing. Can you provide some hook-up details?
Tnx.

pavel-demin, 2017-01-18:
Hi Rick,

Here are the details.

Connections from the [extension connector E1](http://wiki.redpitaya.com/index.php?title=Extension_connectors#Extension_connector_E1) to the MIKROE-506 board:

DIO4_N -> SCK
DIO5_N -> MISO
DIO6_N -> MOSI
DIO7_N -> ADCL

Connections from the [extension connector E2](http://wiki.redpitaya.com/index.php?title=Extension_connectors#Extension_connector_E2) to the MIKROE-506 board:

I2C(SCL) -> SCL
I2C(SDA) -> SDA

The codec is configured via I2C when the program starts on Red Pitaya. Sometimes it helps to restart the program running on Red Pitaya. The most basic test is to check if the codec outputs its clock signals (SCK, ADCL).

Best regards,

Pavel

n1gp, 2017-01-18:
Hi Pavel,

Thank-you, I now have audio out.

I have seen an earlier issue on audio-in and your post of the PSDR settings
will try that next.

Works very nicely thanks again!

closed 2017-01-18 by n1gp