pavel-demin/red-pitaya-notes #452: Build from source sdr_transceiver_emb

Adrianmck, 2017-01-26:
Hi Pavel

Firstly I just want to thank you for a great project.  I'm trying to build your embedded transceiver from source. I have the development machine setup with Vivado 2016.3 and everything compiles and I've created an img file. however when I run a simple example my RedPitaya crashes.  

If I replace the boot.bin with the one form the pre-compiled SD img then everything works. Also if I use the pre-compiled img and put load the new bit file into /dev/xdevcfg then once again the Red Pitaya crashes.

I've also tried reverting the repository to the state it was in on the 2016-06-21 as this is the date on the pre-compiled image, but when I run the make file  I got the following error

ERROR: [BD 41-758] The following clock pins are not connected to a valid clock source: 
/buf_1/OBUF_IN

Any help would be much apprieciated. I'm looking to compile form source as I'm hoping to make changes to how they system works. My ultimate goal is to have the two channels working to record baseband IQ for a separate radio.

Regards

Adrian

pavel-demin, 2017-01-26:
Hi Adrian,

Thanks for your interest in the sdr_transceiver_emb project.

The GNU Radio examples are indeed incompatible with the current version of the FPGA configuration. So, at the moment, reverting to a earlier state of the repository is indeed the easiest solution. I'll try to update the GNU Radio examples in the coming days.

If using GNU Radio isn't a requirement, then here is a link to an example of C code that works with the current version of the FPGA configuration:
https://github.com/pavel-demin/DiscoRedTRX/blob/master/main.c

It can be easily modified to output the I/Q samples that are stored in buffer0:
https://github.com/pavel-demin/DiscoRedTRX/blob/master/main.c#L459

Here is the the last commit with the older FPGA configuration:
https://github.com/pavel-demin/red-pitaya-notes/tree/7e5d50b4e4aca0884c35ff859ce925ea0931aca2

The error message is due to stricter design rules in Vivado 2016.3. I can think of at least two solutions:
 - install Vivado 2016.2 (it was what I used to build the older version)
 - remove buf_0 and buf_1 from sdr_transceiver_emb/block_design.tcl and daisy_* from cfg/ports.tcl

Best regards,

Pavel

Adrianmck, 2017-01-26:
Hi Pavel

Thanks for the quick reply.  I'm going to try  removing buf_0 etc first as the most recent versions have much more features than I need. 

I hoping to remove the DDS etc at the input  as the radio I'm using is outputting analog I/Q. My plan is   to down sample ,that before recording locally. I really like your use of reserved RAM for passing samples form the PL to the CPU. 

I'll let you know how it goes. Is  it better to contact you here or through the Red Pitaya forum?

Thanks again 

Adrian

pavel-demin, 2017-01-26:
> Is it better to contact you here or through the Red Pitaya forum?

Here is perfectly fine.

closed 2017-03-27 by pavel-demin