pavel-demin/red-pitaya-notes #453: Understanding the project tree and using EMIO GPIO

XYEFA, 2017-01-31:
Hello,

I would like to use EMIO GPIO. Where does one begin when opening the project in Vivado (I am using 2016.2)?

Thanks.

pavel-demin, 2017-02-01:
> I would like to use EMIO GPIO. Where does one begin when opening the project in Vivado (I am using 2016.2)?

Sorry, I don't understand your question.

XYEFA, 2017-02-01:
I am trying to create some more GPIO pins and I would like to do this via the EMIO. I want to send this GPIO output to the other Red Pitaya header. I then want to access these new GPIO pins from the Linux kernel (I would need to update the Device Tree).

What I need to know (as the starting point) is where the RedPitaya Vivado project lives so that I can create the new GPIO lines.

pavel-demin, 2017-02-01:
Thanks for more details.

> What I need to know (as the starting point) is where the RedPitaya Vivado project lives so that I can create the new GPIO lines.

Here are the commands to create and to open a project:
```
git clone https://github.com/pavel-demin/red-pitaya-notes
cd red-pitaya-notes
make NAME=led_blinker xpr
vivado tmp/led_blinker.xpr
```

pavel-demin, 2017-02-01:
You can find more details about the project tree and about my toolchain for ZYNQ in my Club Vivado presentation (slides 90-111):
http://www.origin.xilinx.com/publications/products/vivado/club_vivado/2016/einhoven-2016-presentations.zip

XYEFA, 2017-02-01:
Thanks Pavel!

closed 2017-02-12 by pavel-demin