pavel-demin/red-pitaya-notes #454: Optimizations for StemLab10

n1gp, 2017-02-06:
Hi Pavel,

I picked up a RP-10 (Stemlab10) and it seems to work fine
with PSDR & PiHPSDR as-is.

I was just wondering if you can think of any FPGA/Code optimizations
for the 10 -vs- 14 bit paths?

I'd be happy to do any testing if you like.

-Rick / N1GP

pavel-demin, 2017-02-06:
Hi Rick,

Many thanks for the tests. It's good to know that my applications work with the new board.

The most trivial optimization would be, of course, decreasing the bit-width of the data paths. Since the hardware multiplier width is 25 x 18 bits, the bit-width of the data paths could be set to 18 bits and it should be enough for the SDR applications with the 10-bit ADC. But then the applications won't perform well with the 14-bit ADC.

Do you have an idea of a project that would benefit from this kind of optimization?

BTW. Are the markings on the ADC and DAC chips visible on this new board? It would interesting to identify them.

Best regards,

Pavel

n1gp, 2017-02-06:
First I wanted to make sure that with only 10 bits from the ADC/DAC there weren't 4 bits in the data path that could be randomly set (not zero'd out) which might cause some noise or some anomaly, maybe it would even help?

But for optimization I would think if it saved enough resources in the FPGA you may be able to
increase your sdr_receiver_hpsdr from 6 to 7 (or even 8?) rcvrs. That would help the folks who use
it for CW Skimmer to get a couple of extra bands.

I'll try to get the info on the ADC/DAC chips when I get home this evening.

73's

-Rick / N1GP

n1gp, 2017-02-06:
The ADC on my Stemlab10 is a AD9608 BCPZ-125
the DAC seems to be a 14 bit still, AD9767ASTZ (maybe made from left-over stock?)

pavel-demin, 2017-02-10:
> First I wanted to make sure that with only 10 bits from the ADC/DAC there weren't 4 bits in the data path that could be randomly set (not zero'd out) which might cause some noise or some anomaly, maybe it would even help?

According to the [schematics](http://downloads.redpitaya.com/doc/Red_Pitaya_Schematics_STEM_125-10_V1.0.pdf) of the new 125-10 board, the pins corresponding to the four LSB of the 14 bit configuration are connected to GND.

n1gp, 2017-02-10:
That's good!

I didn't know the schematics were published.

closed 2017-02-10 by n1gp