pavel-demin/red-pitaya-notes #455: Debootstrap cannot download Debian

XYEFA, 2017-02-13:
I seem to be having an issue getting Debian - Failed getting release file http://httpredir.debian.org/debian/dists/jessie/Release

How can I resolve this? My network is fine. The messages I get are:

W: Cannot check Release signature; keyring file not available /usr/share/keyrings/debian-archive-keyring.gpg
I: Retrieving Release 
E: Failed getting release file http://httpredir.debian.org/debian/dists/jessie/Release

pavel-demin, 2017-02-13:
Are you running one of the `debian*.sh` scripts? On what kind of machine?

XYEFA, 2017-02-13:
Yes - 

`sudo sh scripts/image.sh scripts/debian-ecosystem.sh red-pitaya-ecosystem-0.95-debian-8.6-armhf.img 3725`

XYEFA, 2017-02-13:
Seems to take forever on my end ... :/ From my Linux box and my Mac Laptop ...

pavel-demin, 2017-02-13:
Looks like it redirects you to a server that is currently down. As a solution, I'd just replace  httpredir.debian.org/debian with an address of a mirror that works for you. Here is a list of mirrors:
https://www.debian.org/mirror/list

XYEFA, 2017-02-13:
Thank you Pavel! I think you can close this.

Regards!

closed 2017-02-13 by pavel-demin

JuiP, 2020-02-01:
can you please tell me how do you replace httpredir.debian.org/debian ? Just a beginner

pavel-demin, 2020-02-01:
> can you please tell me how do you replace httpredir.debian.org/debian ?

I'm afraid that this question is too general for me to answer specifically. Please try to explain what is exactly your problem.

My general answer is to edit the script file you're trying to run and set the `mirror` variable to the URL that works for you.

JuiP, 2020-02-01:
Got it figured out! Thanks :)