pavel-demin/red-pitaya-notes #457: Where did you find instructions on building your debian image?

hmaarrfk, 2017-02-21:
Hey Pavel,

I bought a MicroZed and I'm working on creating a linux distribution for it.

I'm basing my work on your notes as the instructions on Xilinx's website only tell you how to create a [ramdisk image](http://www.wiki.xilinx.com/Getting+Started?responseToken=948833e59b6533fac82de33fb174d74f). Am I missing something from their website? Do you have a good reference that you followed to create your SDCard linux distro?

Thanks again,

Mark

pavel-demin, 2017-02-21:
Hi Mark,

> the instructions on Xilinx's website only tell you how to create a ramdisk image.

There is also PetaLinux:

http://www.wiki.xilinx.com/PetaLinux
http://www.wiki.xilinx.com/PetaLinux+Getting+Started

> Do you have a good reference that you followed to create your SDCard linux distro?

I wouldn't call it a distribution. My SD card images are just custom minimal installations of the Debian and Ubuntu distributions.

All the instructions can be found in the documentation of the distributions that you want to install. Here are a few examples:

https://wiki.debian.org/Debootstrap
https://wiki.debian.org/EmDebian/CrossDebootstrap
https://wiki.ubuntu.com/Base
https://wiki.ubuntu.com/DebootstrapChroot
https://wiki.ubuntu.com/ARM/RootfsFromScratch

There are also plenty examples that can be found with a web search engine:

https://www.google.be/search?q=debootstrap
https://www.google.be/search?q=multistrap

I've just found an example of minimal Debian installation for the MicroZed board:

https://github.com/dasGringuen/debian-microzedboard

For some distributions you can download pre-built root file systems:

http://cdimage.ubuntu.com/ubuntu-base/releases/14.04/release/
http://cdimage.ubuntu.com/ubuntu-base/releases/16.04/release/
https://archlinuxarm.org/about/downloads

Cheers,

Pavel

hmaarrfk, 2017-02-21:
Great thanks!

Yea it is often hard to tell what to follow and what not to follow. I'm following a combination of those guides just to get my head wrapped around what is actually happening.

closed 2017-02-21 by pavel-demin