pavel-demin/red-pitaya-notes #459: HPSDR missing PTT indication back to the PSDR app

ka6s, 2017-02-24:
Pavel,

I've been involved in some discussions with other folks that were noticing PowerSDR not behaving quite right with CW with Break-in set.  To that end I started looking at the server code and comparing the message being sent to PowerSDR versus the Metis protocol.  I can't identify where the server code is sending PTT, Dash, and Dot back up to PowerSDR.

From the Command & Control description for ep6 - found:
Command & Control

NOTE:  Bits 7-3 of C0 form an address that determines how C1-C4 should be decoded. C0 is varied round-robin fashion so that all addresses are sent in sequence. 

C0
0 0 0 0 0 0 0 0
                | | |
                | | + --------- PTT  (1 = active, 0 = inactive), GPIO[23]= Ozy J8-8, Hermes J16-1 
                | +------------ DASH (1 = active, 0 = inactive), GPIO[21]= Ozy J8-6, Hermes J6-2
                +-------------- DOT  (1 = active, 0 = inactive), GPIO[22]= Ozy J8-7, Hermes J6-3 

(Hope this stays put...)

If I understand the code correctly - I don't see this being set any place.

Am I in error?

Thanks,

Steve KA6S

pavel-demin, 2017-02-24:
> I've been involved in some discussions with other folks that were noticing PowerSDR not behaving quite right with CW with Break-in set.

Could you please provide more details about this problem?

I've just checked the latest version (0.94-1451) of the SDR transceiver compatible with HPSDR application and PowerSDR mRX PS 3.3.9. PTT input works. I don't know how PowerSDR is expected to behave when DOT and DASH are activated.

>  I can't identify where the server code is sending PTT, Dash, and Dot back up to PowerSDR.

The code is on line 1012 in sdr-transceiver-hpsdr.c:
https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_hpsdr/server/sdr-transceiver-hpsdr.c#L1012

I've just checked the Hermes code:
http://svn.tapr.org/repos_sdr_hpsdr/trunk/Hermes/Source/Hermes_v3.1/

I see that `Hermes_Tx_fifo_ctrl/clean_PTT_in` is connected to `(clean_PTT_in | CW_PTT)`, where  `CW_PTT` comes from the CW keyer. So, looks like the CW keyer in the Hermes code controls both PTT out and in. In my code, the CW keyer controls only PTT out.

ka6s, 2017-02-24:
> transceiver_hpsdr/server/sdr-transceiver-hpsdr.c#L1012

Pavel,

I'm going to forward a couple of snippets from the conversation that
occurred yesterday - there were over 45 emails in the exchange so I'm going
to try and forward the pieces that defined what Paul is seeing. I have his
permission to forward these to you.

"Hi Doug and Warren,

If I CW TX with my Red Pitaya by first enabling MOX,
the spectrum display changes to the TX display and
on return to RX, the transient in the RX spectrum is
quite small and comes up from the bottom.

OTOH, if I CW TX with break-in, ie no MOX, then the
spectrum display remains in RX and on return to RX,
the transient in the RX spectrum is HUGE and comes down
from the top, ie due to lots of TX signal. It 'appears'
that the RX's are still enabled during TX. This
happens whether MON is enabled or not.

Seems to me this latter behavior is incorrect - ie,
even in break-in, the spectrum should change to TX and
RX should be suppressed.

Please confirm if it is incorrect and if it is different
from other rigs.

If different with the Red Pitaya from other rigs, then
is this a bug that I need to bring to Pavel's attention?

Also with RP, MON is enabled whenever the keyer sidetone
is enabled. Is this the same as other rigs?

Many thanks and 73,
Paul N2PK"

pavel-demin, 2017-02-24:
Moving this to the right issue...

"Below are three PSDR screenshots of CW keyer split TXing with JUST my RP
and RX
antenna preamps powered down.

The first is with MOX and the 2nd is with break-in and the third is with
neither
MOX nor break-in enabled. In all cases, I can hear sidetone from the CODEC
during
keying.

As you can see, the spectrum display is different with MOX.

You can also see a birdie/spur/stray signal at about 1824 kHz that STAYS
about
the same for RX and break-in TX. Sure looks to me like the RP is sending
the same
RX data streams back to PSDR (whatever they are) regardless of TX state.
Should
RP do even that, Warren?

Paul

MOX (same w/ or w/o break-in enabled):

Break-in:

Neither MOX nor Break-in enabled:

pavel-demin, 2017-02-24:
Thanks for these descriptions of the problem. Looks like it's indeed related to my CW keyer code not controlling PTT in. I think that this feature could be easily added.

ka6s, 2017-02-24:
Thank you Pavel - I pass along the encouraging note.

Steve

pavel-demin, 2017-02-24:
I think that [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/e9b352ba38c26e19d9d4bc800542afba4ec78d6e) should fix this problem.

pavel-demin, 2017-03-27:
I've uploaded the new version to the Red Pitaya application marketplace and updated the customized SD image. Closing this issue.

closed 2017-03-27 by pavel-demin