pavel-demin/red-pitaya-notes #460: Debian guest does'nt get IP address from DHCP server located in HostOnly

dnguyen76, 2017-02-24:
Hi Pavel

I had tested this morning new installation you have generated yesterday
I got an small issue as Debian VM does'nt get IP address from DHCP for HostOnly adapter #2 

The insertion in /etc/network/interfaces
auto eth1
    allow-hotplug eth1
    iface eth0 inet dhcp

solve the issue, but it should be a better solution
Kind regards
Daniel

pavel-demin, 2017-02-24:
Hi Daniel,

The idea is/was that network-manager enables both interfaces but, apparently, it only enables eth0.

The only solution that I can think of is to add the following line to the installation script:
```
printf "\nauto eth0\niface eth0 inet dhcp\n\nauto eth1\niface eth1 inet dhcp\n" >> /target/etc/network/interfaces
```

Then network-manager can be probably removed from the list of the installed packages.

Best regards,

Pavel

pavel-demin, 2017-02-24:
The line that works is
```
printf "\nallow-hotplug eth1\niface eth1 inet dhcp\n" >> /target/etc/network/interfaces
```
It's now in the new version of the debian.seed script.

With this line and with network-manager removed, both interfaces are running after the installation.

closed 2017-02-24 by pavel-demin

ka6s, 2017-02-24:
"Below are three PSDR screenshots of CW keyer split TXing with JUST my RP
and RX
antenna preamps powered down.

The first is with MOX and the 2nd is with break-in and the third is with
neither
MOX nor break-in enabled. In all cases, I can hear sidetone from the CODEC
during
keying.

As you can see, the spectrum display is different with MOX.

You can also see a birdie/spur/stray signal at about 1824 kHz that STAYS
about
the same for RX and break-in TX. Sure looks to me like the RP is sending
the same
RX data streams back to PSDR (whatever they are) regardless of TX state.
Should
RP do even that, Warren?

Paul

MOX (same w/ or w/o break-in enabled):

Break-in:

Neither MOX nor Break-in enabled:

ka6s, 2017-02-24:
Pavel,

I'm sorry for the one-sided nature of what i've posted - but I haven't had
a chance to get the permission of the other correspondence yet - I just
want to be above board with all concerned.

As always - your help is most appreciated.  Your initial diagnosis about
the difference with the CW Keyer seems like part of the issue.

Steve