pavel-demin/red-pitaya-notes #464: transfer data into a text file

issmaf44, 2017-03-06:
Dear Pavel,
I would like to transfer digital data  (at the output of an IP compiler) in vivado and send it into a textfile.
Do you have any idea to do that?

Thanks for your help.

pavel-demin, 2017-03-06:
I'm not sure if I understand the meaning of 'digital data (at the output of an IP compiler) in vivado'. Could you please provide a more detailed description or some screenshots?

issmaf44, 2017-03-06:
I am trying to demodulate a signal.  using redpitaya board.
To try this I used a external amplitude modulated signal. The carrier of this signal is of 2MHz frequency.
The sinusoidal signal is transmitted to a complex multiplier through the ADC. In this multiplier, data from the ADC and those from an NCO IP core are mixed. The ouptut of the mixer are then transmitted to a CIC compiler. 
The used CIC decimation have the following parameters:
decimator, 3 stages, differential delay 1, 1 channel, fixed rate 300
input sample frequency : 125 MHz
clock frequency : 125 MHz
input data width : 14 bits
quantization : truncation
output data width : 14 bits
all other parameters are unchanged

I would like to save the data at the ouput of the CIC compiler into a text file.

pavel-demin, 2017-03-06:
Thanks for the details. Now, I better understand the problem. Since the resulting sample rate isn't very high (125/300=0.42 MSPS), both GP and HP buses supports this sample rate.

In some of my projects, I use a combination of fifo_generator and axi_axis_reader (https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver/rx.tcl#L244) that are connected to GP0 (https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver/block_design.tcl#L153).

In some other projects, I use axis_ram_writer connected to HP0 (https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/adc_recorder/block_design.tcl#L217). The adc_recorder project has a very simple C program that prints the ADC samples (https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/adc_recorder/adc-recorder.c).

Here are commands to create Vivado projects:
{{{
git clone https://github.com/pavel-demin/red-pitaya-notes
cd red-pitaya-notes
make NAME=sdr_transceiver xpr
make NAME=adc_recorder xpr
}}}

issmaf44, 2017-03-06:
> master/projects/sdr_transceiver/rx.tcl#L244) that are connected to GP0 (
> transceiver/block_design.tcl#L153).
> master/projects/adc_recorder/block_design.tcl#L217). The adc_recorder
> master/projects/adc_recorder/adc-recorder.c).

I don't really understand, if i see your project, you use an axi 4 stream
FIFO which is combined to an AXi 4 stream Reader and a FIFO Generator.

Can you give me more explanation. Sorry I have a beginner in Vivado design
suite.

Thanks for your help.

pavel-demin, 2017-03-06:
When axi_axis_reader receives a read request from the AXI bus, it reads one entry from FIFO. The source code of the axi_axis_reader module can be found at the following link:
https://github.com/pavel-demin/red-pitaya-notes/blob/master/cores/axi_axis_reader_v1_0/axi_axis_reader.v

closed 2017-03-07 by pavel-demin

parveennisha, 2017-11-10:
Hi Pavel
I am trying to implement the frequency counter project as given in this site http://antonpotocnik.com/?p=519284. While going through the axis _redpitaya_adc.v and axis_redpitaya_dac.v code what is the submodule ODDR. What is the logic of that submodule? From where i can get the whole code?

pavel-demin, 2017-11-10:
@parveennisha, since you question is unrelated to this closed issue, I've created a [new issue](https://github.com/pavel-demin/red-pitaya-notes/issues/626) and copied your question there.

parveennisha, 2017-11-17:
Hi Pavel,
I am trying to implement ( http://antonpotocnik.com/?p=519284 ) this project with some modifications. I took an external clock for counting the frequency through the GPIO pin. I don't know the C coding. Is the C code is generated by launching the SDK in vivado? Can you suggest me what are the changes i need to do in C code for getting the output.
Thanks

pavel-demin, 2017-11-17:
@parveennisha, please create new issues for questions unrelated to this issue that is closed many moths ago.