pavel-demin/red-pitaya-notes #465: Information about Radar project

rogumag, 2017-03-07:
Hi pavel, i see that you add a new project based in an implementation of a Radar. I recreate a this project and executed the server application in a Red Pitaya. I have not see any values in the console. Do you give more information about the utilizatión of this project??. 
Thanks for your work with the Red Pitaya board is very important for me. Best Regards

pavel-demin, 2017-03-07:
This new radar project is a work in progress. Last week, I've put together some pieces from my other projects to have an initial FPGA configuration for the [PiRadar project](https://www.scivision.co/pi-radar/), The FPGA code builds but it's untested and there could be some bugs/typos. The C code has some parts missing.

rogumag, 2017-03-07:
ok, thank you for your faster response. I will waiting for your updates. I will review your PiRadar project.

closed 2017-03-30 by rogumag