pavel-demin/red-pitaya-notes #469: Modifying the device tree and loading a kernel mod

Yatekii, 2017-03-27:
Hi!

So I have built my custom ADC triggering/recording unit upon your sources, which are awesome!
Sadly I don't always fully understand everything since it's very loosely documented :)
I have all the basic understanding on how the device tree, kernel drivers and communication with AXI works.

I also wrote a kernel module and the configured the entire FPGA side. Only the device tree part was done for me last time.

Now on this new board I need a way to get my kernel mod being loaded, but I really can't figure where in the thousands of dts files I need to put my regs.
I found this patch: https://github.com/pavel-demin/red-pitaya-notes/blob/master/patches/devicetree.patch. There you add the ADC registers.

```
&adc {
	xlnx,channels {
		#address-cells = <1>;
		#size-cells = <0>;
		channel@0 {
			reg = <0>;
		};
		channel@1 {
			reg = <1>;
		};
		channel@2 {
			reg = <2>;
		};
		channel@9 {
			reg = <9>;
		};
		channel@10 {
			reg = <10>;
		};
	};
};
```

But I honestly don't fully understand :S Those numbers can't possibly be memory locations where the ADC data is stored?

So where exactly do I put my block of the likes:

```
xillybus_0: xillybus@50000000 {
      compatible = "xlnx,xillybus-1.00.a";
      reg = < 0x50000000 0x1000 >;
      interrupts = < 0 59 1 >;
      interrupt-parent = <&gic>;
      xlnx,max-burst-len = <0x10>;
      xlnx,native-data-width = <0x20>;
      xlnx,slv-awidth = <0x20>;
      xlnx,slv-dwidth = <0x20>;
      xlnx,use-wstrb = <0x1>;
    } ;
```

(just an example, would be the one from my kernel mod.)

Also: I know the ZYNQ has 500MB onboard memory and 4GB external DDR3. I guess it's no big deal interfacing them both via AXI. But what I saw in the dts files in this repo, I feel like the linux memory resides in the 4GB DDR3, located at 0x0?
Also the size of memory annotated in the dts is always huuuge. Why does it not tell the actual size?
I guess I could easily constrain the Linux to the onboard mem at 0x40000000 and store my ADC data to 0x0, right?

I am very grateful that you are keeping this repo alive; it litterally saved me.
Thanks in advance for any answers.

Best wishes
Yatekii

pavel-demin, 2017-03-27:
> So I have built my custom ADC triggering/recording unit upon your sources, which are awesome!

Congratulations! I'm glad that my notes helped you.

> So where exactly do I put my block of the likes:

No idea. I don't use any custom Linux modules in my projects.

> I know the ZYNQ has 500MB onboard memory and 4GB external DDR3.

I'm afraid that your information isn't correct. The Red Pitaya board has only 512 MB of DDR3. The corresponding number of bytes in the hexadecimal representation is 0x20000000.

> Also the size of memory annotated in the dts is always huuuge. Why does it not tell the actual size?

I don't understand this part. I didn't see any wrong numbers in the dts files.

Yatekii, 2017-03-27:
Wow, you also answer really quick! Thanks!
Ok well I guess then I have to figure myself where to patch the dts files :) I mean it should be doable ;)
Hmm ok then my information sadly is incorrect. I assumed it has to be 4 GB since the Red Pitaya project to me is kinda a mess (no offense to anyone ..) and I can't find the exact numbers. But in your adc_test you had memory mapped to DDR3 and on the Kickstarter they wrote that they would upgrade to 4GB DDR3 if there is more than 150k$ in pledges which it had. So I assumed it has to be like that ... kinda naive of me.

Ok I guess I misread the numbers on the memory size like the noob I am.

Can you tell me where I get RELIABLE information on the HW? :S Does the board have 1GB in total (512MB onboard, 512MB DDR3) or 512MB?

So I guess for your ADC modules you constrained the Linux to 512MB and put your ADC data to the memory regions above?

Side question: should I open an issue for every question I have and is this okay? :S (I hope there wont be too many anymore, but I am sure there is one or the other coming up soon(TM))

Thanks again for the great help!

pavel-demin, 2017-03-27:
BTW. Since you've copied the cores from this repository to [alpenwasser/pitaya](https://github.com/alpenwasser/pitaya/tree/master/firmware/fpga/cores), please don't forget to copy the copyright and license notices from  https://github.com/pavel-demin/red-pitaya-notes/blob/master/LICENSE

Yatekii, 2017-03-27:
Oh, sure, I am sorry I forgot that ... I for now added you as vendor on any of the cores but forgot the license. I didn't mean any troubles :(

pavel-demin, 2017-03-27:
> Kickstarter they wrote that they would upgrade to 4GB DDR3 if there is more than 150k$ in pledges which it had.

Just checked the Kickstarter page. It's indeed a little bit confusing. The amount of memory is expressed in gigabits (Gb) not in gigabytes (GB) or megabytes (MB). The numbers should be divided by eight to convert them to megabytes: 1 Gb = 256 MB and 4 Gb = 512 MB.

> Can you tell me where I get RELIABLE information on the HW?

I think that the most up-to-date information is in the Red Pitaya documentation:
http://redpitaya.readthedocs.io/en/latest/doc/developerGuide/125-10/vs.html

Some tables should be scrolled horizontally to see all the numbers :-)

> Does the board have 1GB in total (512MB onboard, 512MB DDR3) or 512MB?

As far as I know, on-board and DDR3 are the same thing in case of this board. The 125-14 (original Red Pitaya) board has 512 MB of DDR3 and the new 125-10 board has 256 MB of DDR3.

> Side question: should I open an issue for every question I have and is this okay?

Better keep all the questions in this single issue.

> So I guess for your ADC modules you constrained the Linux to 512MB and put your ADC data to the memory regions above?

It depends on what project you're looking at. I have just a few project there the ADC samples are copied to DDR3. In these project Linux uses 480 MB of DDR3 and the ADC samples are copied to the last 32 MB of DDR3.

The 480 MB are set by the following lines:
https://github.com/pavel-demin/red-pitaya-notes/blob/master/patches/devicetree.patch#L25-L26
https://github.com/pavel-demin/red-pitaya-notes/blob/master/patches/u-boot-xlnx-xilinx-v2016.4.patch#L21

In most other projects, I use small FIFO buffers (FPGA memory blocks) that are accessed from the CPU via the AXI bus. The addresses of these FIFO buffers could be any numbers outside of the DDR3 addresses.

closed 2017-04-04 by pavel-demin