pavel-demin/red-pitaya-notes #470: Embedded SDR out of tree blocks

zhudsonSM, 2017-04-04:
I have some out of tree blocks in C++ that I would like to use on the Red Pitaya. Is there a way to move them over without having to compile them on the Red Pitaya?

pavel-demin, 2017-04-04:
You can cross-compile them on a PC.

zhudsonSM, 2017-04-05:
Thanks, that makes good sense. I am trying to follow this from the wiki: https://wiki.gnuradio.org/index.php/Cross_compile_an_OOT_and_install_on_target (is that correct?) and am getting an error from cmake saying "The C++ compiler "/usr/bin/c++" is not able to compile a simple test program"

pavel-demin, 2017-04-05:
The wiki page looks like a good starting point. I never used cmake for armhf cross compilation. Only used it once for win32 cross-compilation. Correct `CMAKE_TOOLCHAIN_FILE` was essential.

I've just found a toolchain file for armhf cross-compilation that looks reasonable:
https://github.com/zyga/cmake-toolchains/blob/master/Toolchain-Ubuntu-gnueabihf.cmake

zhudsonSM, 2017-04-05:
I get the same problem with that toolchain file. I will go ask more generally on Stack Overflow 

What method would you suggest for cross compiling if you don't use cmake?

pavel-demin, 2017-04-06:
> What method would you suggest for cross compiling if you don't use cmake?

I never cross-compiled GNU Radio blocks. Can't suggest anything.

closed 2017-04-06 by pavel-demin