pavel-demin/red-pitaya-notes #471: pre-build red-pitaya images on KVM/qemu-system-arm?

berni421, 2017-04-05:
I am trying to load the actual sd image into a virtual machine. I hope to run it as a VM on my x64 laptop for familiarization.

I have the red-pitaya-ecosystem-0.95-debian-8.6-armhf-20161219.img  but sadly I cant find the kernel, or initrd image in the file. I have used virt-builder --get-kernel and a look round using kpartx. 

Where are the boot files please? 

There is also an option for  a DTB. Is this needed? 

Or perhaps I should also ask if this is just something that just wont work?

Huge thanks

Berni

pavel-demin, 2017-04-05:
> Where are the boot files please?

They are on the first partition. Here is a list of files on the first partition:
```
boot.bin
devicetree.dtb
uEnv.txt
uImage
```
The image can be mount using `losetup` and `mount` as it's done in [image.sh](https://github.com/pavel-demin/red-pitaya-notes/blob/master/scripts/image.sh) and [debian.sh](https://github.com/pavel-demin/red-pitaya-notes/blob/master/scripts/debian.sh)

> I hope to run it as a VM on my x64 laptop for familiarization.
> Or perhaps I should also ask if this is just something that just wont work?

If it's just for familiarization, then it could be done without VM.  It's possible to use `qemu-arm-static` and to `chroot` to the second SD card image partition as it's done in [debian.sh](https://github.com/pavel-demin/red-pitaya-notes/blob/master/scripts/debian.sh).

closed 2017-04-05 by pavel-demin

berni421, 2017-04-06:
Hmm. Thanks for confirming the files.

It seems kvm uses slightly different terminology. A kernel path, initrd path, dtb path, and kernel args. dtb file is obvious. The rest suggest to me kvm is not going to work. Also I can only see armv7l  architecture.