pavel-demin/red-pitaya-notes #472: transmit-wspr-message doesn't work

anax1000, 2017-04-07:
Hello,

I've dd-ed the red-pitaya-wspr-debian-8.6-armhf-20161219.img to a SD-Card.
Logging in via ssh to the red-pitaya works.
WSPR-receptions works too very well..
But transmission doesn't.
I have edited the transmit-wspr-message.cfg,
but if i start transmit-wspr.sh which should start /root/transmit-wspr-message,
/root/transmit-wspr-message exits immediately without any error.
What i'am doing wrong?

Thank you very much for your help.

Alex (DD1ALX)

pavel-demin, 2017-04-07:
> but if i start transmit-wspr.sh which should start /root/transmit-wspr-message,
/root/transmit-wspr-message exits immediately without any error.
> What i'am doing wrong?

According to you description, the program works as it should. It send all the information to FPGA and exists.

To enable the scheduled transmission, the second line in wspr.cron should be uncommented.

anax1000, 2017-04-07:
Hello Pavel,
i'am very sorry, it really works :-)
I was confused, because there is no process on the
linux-side after starting the transmit.

Thank you very much!

Alex

closed 2017-04-07 by anax1000

pavel-demin, 2017-04-07:
Thank you for testing it and for confirming that it works.