pavel-demin/red-pitaya-notes #476: GPIO access

zhudsonSM, 2017-05-08:
I am using the gnuradio-emb project to create an embedded transmitter. Is there a way to be able to use the GPIO pins on the RedPitaya also? I understand they are connected to the FPGA but can't find any decent documentation on accessing them other than over SCPI. I could also use the LEDs if they are easier to control. 

Thank you

pavel-demin, 2017-05-09:
I think that [gpio-output.c](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_wspr/gpio-output.c) from the sdr_transceiver_wspr project can be also used with the sdr_transceiver_emb project to control the DIO*_P pins of the [extension connector E1](http://redpitaya.readthedocs.io/en/latest/doc/developerGuide/125-14/extent.html#extension-connector-e1).

zhudsonSM, 2017-05-12:
Thank you, I am now able to toggle IO. However as soon as I run the GNURadio part it resets to having only  GPIO[0] on. It looks like it is using that same memory location for the self.cfg struct in red_pitaya_emb.py specifically it looks like it is using them for rates and frequency. Is this correct? Is there any way to be able to use the IO pins while the radio is on?

pavel-demin, 2017-05-15:
> It looks like it is using that same memory location for the self.cfg struct in red_pitaya_emb.py specifically it looks like it is using them for rates and frequency. Is this correct?

Yes, it's correct. The code on lines 78-83 and 95-100 in red_pitaya_emb.py sets and resets the GPIO pins:

https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_emb/gnuradio/red_pitaya_emb.py#L78-L83
https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_emb/gnuradio/red_pitaya_emb.py#L95-L100

>  Is there any way to be able to use the IO pins while the radio is on?

I think that removing the lines 78-83 and 95-100 from red_pitaya_emb.py should solve the problem.

zhudsonSM, 2017-05-15:
Thank you for your help with this. Could you help me to understand what the original intention of writing to the IO lines is? Was it for IO to some PTT circuit that you had it hooked up to for your testing or is there some other use of that memory location? Also I have been unable to find any documentation pointing to how those specific memory addresses are tied to GPIO. Is that just something specific to the FPGA image that you have or is there some documentation that I could learn about that?

pavel-demin, 2017-05-15:
> Could you help me to understand what the original intention of writing to the IO lines is? Was it for IO to some PTT circuit that you had it hooked up to for your testing or is there some other use of that memory location?

The idea was to control a PTT circuit from GNU Radio.

> Also I have been unable to find any documentation pointing to how those specific memory addresses are tied to GPIO. Is that just something specific to the FPGA image that you have or is there some documentation that I could learn about that?

It's specific to the FPGA image. Connection between  the GPIO pins and the configuration register is defined in [sdr_transceiver_emb/block_design.tcl](https://github.com/pavel-demin/red-pitaya-notes/blob/89405607747a0a20917f662cf37a6f1f9ac8b9f9/projects/sdr_transceiver_emb/block_design.tcl#L114-L128).

closed 2017-05-25 by pavel-demin