pavel-demin/red-pitaya-notes #478: Error writing files from WSPR Transceiver

dcmk1mr2, 2017-05-12:
I downloaded the software and have tried several SD cards and variations on the instructions.  
The error message's are variations on this:

root@red-pitaya:~# ./decode-wspr.sh
Fri May 12 18:42:42 CEST 2017
Sleeping ...
Fri May 12 18:43:00 CEST 2017
Recording ...
Killed write-c2-files(19063) with signal 15
Terminated
Decoding ...
Cannot open data file 'wspr_*_170512_1643.c2'
root@red-pitaya:~#

Thanks very much,
Martin W6MRR

pavel-demin, 2017-05-12:
I think that the problem is that you're running `decode-wspr.sh` from the command line. Normally, `decode-wspr.sh` is started automatically by cron as it's described in the software section: 
http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-wspr/#software

The cron configuration is in [wspr.cron](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_wspr/wspr.cron).

dcmk1mr2, 2017-05-13:
> transceiver-wspr/#software

Hi Pavel,

Thanks so much! Just amazing software.

I will re-read the instructions next time.

73,

Martin
W6MRR

closed 2017-05-13 by pavel-demin