pavel-demin/red-pitaya-notes #479: ADC IP output format

Yatekii, 2017-05-17:
Hi Pavel!

I have been working with the Red Pitaya for a few weeks now and pretty much everything works great now. Thanks for all the work you have done!

But one thing seems to be off:
The ADC is 14 bit. And you convert it somehow to 16bits. It feels like you sign extend and then invert bits but I don't quite understand what the point is. The problem is that my own core was made to handle only unsigned values and now I have some issues with the format.
So can you tell me what the format of the ADC 2 IP actually is? Is it signed 16bit?

Thanks!

pavel-demin, 2017-05-17:
> It feels like you sign extend and then invert bits but I don't quite understand what the point is.

Yes, the the ADC samples coming from the ADC IP core are sign extended and inverted.

The signal at the ADC input is bipolar and inverted. Then ADC outputs samples in the two's complement format. Since the DSP IP cores (CIC, FIR, Complex Multiplier) provided by Xilinx work with the two's complement format, my IP cores also use this format.

The inversion is needed to have the same signal polarity as the board's input.

> So can you tell me what the format of the ADC 2 IP actually is? Is it signed 16bit?

Yes, the format is signed. The number of bits is controlled by the ADC_DATA_WIDTH parameters. If ADC_DATA_WIDTH is 16, then the format is signed 16-bit.

Yatekii, 2017-05-17:
Oh, so I need to read about bipolar :) Because I didn't 100% understand the conversion. Thank you!
So I will work with those signed 16-bit, thank you very much!

pavel-demin, 2017-05-17:
Correction to my previous comment.

I've just reread [axis_red_pitaya_adc.v](https://github.com/pavel-demin/red-pitaya-notes/blob/master/cores/axis_red_pitaya_adc_v2_0/axis_red_pitaya_adc.v) and it looks like ADC outputs samples in the offset binary format (not two's complement).

So, my ADC IP core converts offset binary to two's complement, extends sign and inverts the ADC samples.

Yatekii, 2017-05-17:
Ok, why does it invert? I don't quite get it ... The problem is that when I read a sine now this works fine until I hit a boundary and it should start clipping but it doesn't clip but wraps around. This is not a behavior I like. And I guess I can't even really fix it without changing your core, right?

Because when I plot the sine I like to have it clipped at the top and not wrap around to the bottom, even tho for CIC filters for example, a wraparound is required.

If this question is too off topic, no problem, but I greatly appreciate your help :)

pavel-demin, 2017-05-17:
> Ok, why does it invert?

There is an inverting amplifier somewhere between the SMA connector and the ADC input. My ADC IP core inverts the ADC samples to have the same signal polarity as at the SMA connector.  It's not strictly necessary. It's just for my convenience. 

> I hit a boundary and it should start clipping but it doesn't clip but wraps around.

It's possible that there is a problem with my inversion code. I can't check it right now but I'll check it later.

Here is how I think my IP core should behave:

- top clipping (+1.1 V) -> inverting amplifier (-1.1 V) -> ADC (`00 0000 0000 0000`) -> IP core (`0001 1111 1111 1111`) -> signed 16-bit int (8191)

- bottom clipping (-1.1 V) -> inverting amplifier (+1.1 V) -> ADC (`11 1111 1111 1111`) -> IP core (`1110 0000 0000 0000`) -> signed 16-bit int (-8192)

BTW, do you have an illustration of the wrapped signal?

Yatekii, 2017-05-17:
So to understand a bit better I had a look at what [Wikipedia](https://en.wikipedia.org/wiki/Offset_binary) says to Offset Binary.
So as I understand you would have to extend and invert the sign and leave the actual value untouched, but you do the opposite in your code. And since I know that your code should work since you are already using it, I am confused now :P

pavel-demin, 2017-05-17:
> So as I understand you would have to extend and invert the sign and leave the actual value untouched

I did it like that in earlier versions of the IP core. Here is the commit that added the inversion: https://github.com/pavel-demin/red-pitaya-notes/commit/a2ecbfbda90c7cf8b57d90f597b7f93e9577c035.

The idea of the inversion was not mine. I borrowed it from the original Red Pitaya code:
https://github.com/RedPitaya/RedPitaya/blob/v0.95-1/fpga/rtl/red_pitaya_top.v#L350-L352

Yatekii, 2017-05-17:
Ok this is very interesting. No hurry ;) 

Hmm I agree that the conversion should work. At least for the few cases I checked the math is right.
But then again I do not understand why my signal wraps and doesn't clip. Maybe I have some silly mistake still. I will check and update in here afterwards :)

Yatekii, 2017-05-17:
Ok, I also had a look at the ADC datasheet and your conversion seems indeed right. So the funny question would be why I experience that wrap around instead of a simple clip, because I do not temper with the data lol ... I just write it down.
[image]

[image]

[image]

I am 100% positive that it is my mistake but I don't know why ...

I also tried adding an Offset of 2^13 to the signed 14bit value to make it binary offset (unsigned again) with the same outcome (I only transmit the lower 14bits (the upper 2 are duplicates anyways (sign extend)).

Maybe I am just really dumb ...

pavel-demin, 2017-05-17:
Thanks for the plots and for the data files. I've checked the samples in [middle_sine_wrap.txt](https://github.com/alpenwasser/pitaya/blob/master/verification/2017-05-03--first-success/middle_sine_wrap.txt). The values `3F3C 3F32 3F32 3F1B` are 14-bit signed converted to 16-bit without propagating the sign bit to the first two most significant bits. So, the plot with the wrapping signal shows why it's important to correctly propagate the sign bit.

Yatekii, 2017-05-18:
Thanks for your answer!
I suspected that due to my calculations. But I made an IP which translates the 16bit signed to 16bit unsigned. I also made a testbench which verifies the functionality 100%. Sadly the results stayed the same after flashing the bitstream. So I was debugging the entire afternoon but couldn't find a cause ... maybe "reflashing" a bitstream is not possible for certain functionality, who knows ...

Sorry for cluttering the repo with my own issues :(

closed 2017-05-25 by pavel-demin