pavel-demin/red-pitaya-notes #482: Behaviour of MCPA on STEMLAB10

sefffal, 2017-06-13:
Hi Pavel,
Thanks for all the hard work you've put into these applications.

I'm using your MCPHA on a STEMLAB-10 and I wonder if you can explain some of the results I'm seeing. The data from the ADC on the STEMLAB 10 has the bottom four bits connected to ground to pad the data out to 14 bits. Therefore, in the MCPHA I expected to see channels that weren't a multiple of 16 to be empty. Instead, I'm seeing that these bins are filled as much as the bins that *are* multiples of 16. 

Are you performing some kind of interpolation of the pulses to determine the height?
Also, what is the purpose of the decimation factor in your interface? Does it affect the number of bins in the histogram, or is it decimation applied to the data from the ADC before processing?

I know you wrote the software for the 14-bit ADC, but I would appreciate any insight you could provide.
Thank you,
-Will

pavel-demin, 2017-06-13:
Hi Will,

The ADC samples are processed by the CIC and FIR filters as shown on the diagram at [this link](http://pavel-demin.github.io/red-pitaya-notes/mcpha/#hardware). The filters smooth out the steps in the 10-bit data.

The decimation factor is a parameter of the CIC filter. It doesn't affect the number of bins. Its purpose is to control the pulse duration as seen by the pulse height analyzer. For an optimal resolution, the pulse must not be too wide nor too narrow.

Best regards,

Pavel

sefffal, 2017-06-15:
HI Pavel, thanks for the information.

closed 2017-06-15 by sefffal