pavel-demin/red-pitaya-notes #490: Questions about SDR applications for Red Pitaya

pavel-demin, 2017-07-09:
From @stevec5000:

> Pavel can you tell me how to get the Red Pitaya to work with PowerSDR?
> In your Red Pitaya Notes, SDR Receiver, you say:"The implementation of the SDR receiver is quite straightforward" and "The I/Q data is transmitted via TCP to the SDR programs such as SDR# and HDSDR".
> but it doesn't seem to be any of these! I know of no way to get it working with SDR# so I've been trying to get HDSDR to work with the Red Pitaya SDR app but haven't been able to find any instructions or anyone who will admit to knowing anything about it!
> There are video showing it working on Youtube but they seem to all be fakes, like the ones for quadcopters that can't really fly such as the Lily drone.
> As for HDSDR, I've installed the program in Windows 10 but when I start the Red Pitaya app the SDR program can't seem to be able to find the Red Pitaya.
> The hardest part about this is finding anyone who will admit to knowing anything about it!
> When I tried to run the Red Pitaya SDR radio app it says: "HAMlab is now running in SDR mode. To use it as a radio please connect to it with PowerSDR HAMlab edition. That can be downloaded from this link." and they give a link to the download site at Flex Radio.
> So I installed Power SDR Open HPSDR like it wanted on the PC but when I try to run it by clicking the Power button I get a message saying:
> "No Charly 25/Hamlab device detected on the network, Please power up your Charly 25 / Hamlab before using the autosensing mode!"
> Do you know what this means, what a Hamlab is or what a Charly 25 is or what autosensing mode is?
> I have an SDR USB dongle running with SDR Sharp and that was easy compared to this!
> 
> In Power SDR/OPENHPSDR under Setup, Hardware Config., when I click the Hamlab option it has a window for the RedPitaya IP and Mac Address but it's blank and there's no way to enter the settings into it!
> The program, which was downloaded from Flex Radio Systems, like it says to do in the Red Pitaya SDR app, is ver. 2017-02-18 Commit 15680ad.
> Did I get a bad version of the program that doesn't work?
> How can I get PowerSDR to recognize and work with the Red Pitaya?
> Thanks.

pavel-demin, 2017-07-09:
> Pavel can you tell me how to get the Red Pitaya to work with PowerSDR?

My 'SDR transceiver compatible with HPSDR' application for Red Pitaya is known to work with a version of PowerSDR provided by the High Performance Software Defined Radio (HPSDR) project. Here is a link to my notes about this application for Red Pitaya:

http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-hpsdr

PowerSDR provided by the High Performance Software Defined Radio (HPSDR) project can be downloaded from the following link:

https://github.com/TAPR/OpenHPSDR-PowerSDR/releases

The very simple instructions on how to get the Red Pitaya board to work with this particular version of PowerSDR can be found in the 'Getting started' section of my notes:

http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-hpsdr/#getting-started

pavel-demin, 2017-07-09:
> In your Red Pitaya Notes, SDR Receiver, you say: "The implementation of the SDR receiver is quite straightforward" and "The I/Q data is transmitted via TCP to the SDR programs such as SDR# and [...]

Have you tried to follow the instructions from the 'Getting started' section of my notes about the 'SDR receiver' application for Red Pitaya? Here is a direct link to this section:

http://pavel-demin.github.io/red-pitaya-notes/sdr-receiver/#getting-started

The 'SDR receiver' application for Red Pitaya requires a version of SDR# that supports ExtIO plugins. There is a link to a version of SDR# that works with the 'SDR receiver' application for Red Pitaya in the 'Getting started' section of my notes about the 'SDR receiver' application for Red Pitaya. Here is a direct link to this version of SDR#:

https://www.dropbox.com/sh/5fy49wae6xwxa8a/AAAdAcU238cppWziK4xPRIADa/sdr/sdrsharp_v1.0.0.1361_with_plugins.zip?dl=1

If I'm not mistaken, 1.0.0.1361 was the last version of SDR# that supported ExtIO plugins.

pavel-demin, 2017-07-09:
> When I tried to run the Red Pitaya SDR radio app it says: "HAMlab is now running in SDR mode. To use it as a radio please connect to it with PowerSDR HAMlab edition. That can be downloaded from this link."

I don't know much about HAMlab. So, please contact the HAMlab developers with the HAMlab related questions.

ted051, 2017-07-09:
"Charly25" look here: 
https://www.darc.de/cqdlonline/2017/07/html5.html
questions to HAMLab: 
http://hamlab.redpitaya.com/
Any hints about installation and related problems you can find here:
http://forum.cq-nrw.de/viewforum.php?f=40

stevec5000, 2017-07-09:
Thanks for the reply.  It turns out that most of the Getting Started notes are irrelevant except for the 2 steps that say to Download pre-built ExtIO plug-in for SDR# and HDSDR and Copy ExtIO_RedPitaya.dll into the SDR# or HDSDR installation directory.
That's all that was needed to get HDSDR working except to plug in a mic. for some reason so now it will play.

I haven't been able to get SDR# to work even after coping the file into it's folder.  It can see the R.P. now but when I try to play it nothing happens.  I'd much rather use SDR# since HDSDR is loaded down with all sorts of bloatware and options that aren't needed. 
Do you know how to get SDR# to work?  Thanks.

pavel-demin, 2017-07-10:
>  It turns out that most of the Getting Started notes are irrelevant except for the 2 steps that say to Download pre-built ExtIO plug-in for SDR# and HDSDR and Copy ExtIO_RedPitaya.dll into the SDR# or HDSDR installation directory.

> Do you know how to get SDR# to work?

I'm afraid that the information that you provide isn't enough for me to understand how exactly your Red Pitaya is configured. If you aren't following the steps from the 'Getting started' section, then please provided a detailed description of which of my applications you're using and how exactly you run it on Red Pitaya. The more details, the better.

stevec5000, 2017-07-10:
Pavel, it is not configured in any special way.  All I do is turn it on and run the SDR radio app!
I copied the ExtIO_RedPitaya.dll filw into the SDR# directory so it can find the R.P. now but when I try to Start it nothing plays.
That procedure worked for PowerSDR and it will play but SDR# does not play.
What more do I need to do to SDR# to make it play?  Thanks.

stevec5000, 2017-07-10:
Ted051, thanks for the link to the German forum.

ted051, 2017-07-10:
Hello Steve,
you must go to the "Configure" button (third from left), then insert the correct ip- address of your Red Pitaya. If you don't know it, take the program:
http://www.advanced-ip-scanner.com/
Wolfgang

stevec5000, 2017-07-10:
Ted, I assume you are referring to the SDR# program?  Yes I know about the IP address, I put in the IP address but it still does nothing.

ted051, 2017-07-10:
You wrote: "That procedure worked for PowerSDR and it will play but SDR# does not play."
Did you mean "HDSDR" instead of "PowerSDR"??
The firmware for PowerSDR and SDR# / HDSDR isn't the same!

pavel-demin, 2017-07-10:
> Pavel, it is not configured in any special way. All I do is turn it on and run the SDR radio app!

I'm afraid that it's still unclear for me how your Red Pitaya is configured. Here is a short list of points that would help me to understand the problem:

 - What OS image is written to the micro SD card?
 - What is the exact title of the 'SDR app' and how it was installed?

Reading your description of the problem, I'd guess that you're using a SDR app and a ExtIO_RedPitaya.dll library that aren't compatible. But it's only a guess because I don't even understand what do you mean by 'SDR app'. If by the 'SDR app' you mean the 'SDR transceiver' application installed via the Red Pitaya application market place, then the correct .dll is ExtIO_RedPitaya_TRX.dll (with '_TRX' at the end of the file name).

stevec5000, 2017-07-10:
Pavel, I have no idea if they are compatible or not since I'm a network administrator, not a programmer but I was using the ones mentioned in your article.  I haven't seen ExtIO_RedPitaya_TRX.dll mentioned before, where do I find that?

stevec5000, 2017-07-10:
I found the file ExtIO_RedPitaya_TRX.dll and copied it into the SDR# folder but it still won't play!

DH1KLM, 2017-07-11:
What Hardware are you using?
Is it a Hamlab or a STEMlab (red pitaya)?
Do you habe any other Hardware connected to it, linke lpf and such?
Are you running the original red pitaya image or pavels image?
For the first steps i recommend using the standard image from red pitaya.

If you got the C25/Hamlab message i assume you are running the wrong image.
Dont usw powersdr from flex radio.
Use this one. https://github.com/TAPR/OpenHPSDR-PowerSDR/releases

pavel-demin, 2017-07-11:
@stevec5000, since you're insisting on not answering my questions, then I don't know how to help you. If you're still interested to get some help from this issue, then please try two answer my two simple questions from my previous comment.

stevec5000, 2017-07-12:
Why do you insist on being so difficult?
The OS image is OS 0.97-396 from the website, naturally!
The SDR app is called SDR HPSDR!
I have no idea how they installed it!
I installed one called Radiobox but I can't make sense out of that at all and tried to contact the developers to ask what it does but they seem to be in hiding.

pavel-demin, 2017-07-12:
> Why do you insist on being so difficult?

Oh, I greatly apologize for my extrasensory perception not being good enough to read your mind remotely and to get the answers without making you to type them.

> The OS image is OS 0.97-396 from the website, naturally!
> The SDR app is called SDR HPSDR!
> I have no idea how they installed it!

Thanks for these answers. Now, I start to understand how your Red Pitaya is configured.

According to a recent comment on the Red Pitaya forum, the preinstalled SDR application should be removed:
http://forum.redpitaya.com/viewtopic.php?t=2032&p=7694#p7694
> please remove this application and install SDR app from marketplace. SDR app in BETA OS release was there for beta testers only, but we forgot to put a note about that.

So, I can recommend the following steps:

 - start the application marketplace via the Red Pitaya web interface
 - remove all the preinstalled SDR applications
 - install 'SDR transceiver'
 - install 'SDR transceiver compatible with HPSDR'

'SDR transceiver' can be used with SDR# and HDSDR.

'SDR transceiver compatible with HPSDR' can be used with PowerSDR mRX PS and other programs that support the HPSDR/Metis communication protocol.

stevec5000, 2017-07-12:
>Oh, I greatly apologize for my extrasensory perception not being good enough to read your mind remotely and to get the answers without making you to type them.

I just took the item straight out of the box and tried to use it so you should know how it's setup more than I do since I never used one before!

I can't install a new SDR app and there is no way to remove the pre-installed apps!  The app store is not responding since installing the last Beta image, 0.97-396 and seems to be off line!
Why is a product that's been out for a couple of years still using Buggy Beta software?

DH1KLM, 2017-07-12:
@stevec5000 
Why do you have installed a Beta Image? 
Beta Software will ever exist beside a stable release candidate.

I recommend to have a look on this. http://redpitaya.readthedocs.io/en/latest/
And for installing a stable version. http://redpitaya.readthedocs.io/en/latest/quickStart/SDcard/SDcard.html

stevec5000, 2017-07-12:
I had to update to Beta image 0.97-396 to get the Logic Analyzer to work!
Is there something newer and better?

DH1KLM, 2017-07-12:
I do not have any experience with the Logic Analyzer since i dont use it.
I assume when running the latest stable version the Logic Analyzer should also
run like SDR Transceiver and all other "Apps" in the Red Pitaya Marketplace.

stevec5000, 2017-07-12:
No, the logic analyzer said I needed a product key to unlock it which is why I was told to use 0.97-396.
This is the most half-baked product I've ever seen!  Doesn't anyone at Red Pitaya know what they are doing or how to fix all the bugs?

ted051, 2017-07-12:
Hi Steve,
I think, Pavel Demin is the false address for your troubles!
Have a look at 
http://forum.redpitaya.com/
or send your problems to 
https://redpitaya.com/

stevec5000, 2017-07-12:
The Red Pitaya Forum hasn't been any help at all.  At least Pavel Demin seems to know something about it and has a blog that tells how to get parts of it working that is helpful even if some of it is wrong and hard to understand.

ka6s, 2017-07-12:
Steve,

You need to re-think your approach here.  First - Pavel is NOT the company
that makes the Red Pitaya - he is a user of their product, just like us.
Next - he has made all of these packages available and tries to help in the
spirit of good will. He owes us NOTHING!  He isn't being paid for this
stuff! I personally have had several interactions in the past with Pavel
that were quite fruitful and pleasant.  So maybe it isn't him??   Your
expectations aren't aligned with reality!

Basic point - he asked you specific questions so he could help you -
multiple times - and you didn't answer his questions - so I quite
understand his latest correspondence to you.  You are making assumptions as
to what he knows about the state of your hardware - he doesn't know - that
is why he asked!

So if you cooperate - and give him the info he asks - maybe he can help,
but please remember- he doesn't owe ANY response to anyone!  There is
simply no obligation he has to respond.

Steve Wilson

stevec5000, 2017-07-12:
Not aligned with reality?  You mean expecting it to work?  Like I said, Pavel's blogs are confusing and somewhat wrong but at least it's some help, unlike anyone from Red Pitaya!  They don't even reply to any questions on their forum!  To be fair the CEO did tell me their friend who used to run the forum died but still that's no way to run a business and they should find someone else instead of just abandoning all their customers!

pavel-demin, 2017-07-12:
> I can't install a new SDR app and there is no way to remove the pre-installed apps! The app store is not responding since installing the last Beta image, 0.97-396 and seems to be off line!

These are bad news (if you're sure that it's not a problem with Red Pitaya having access to the Internet). I never tested the 0.97 versions of the SD card images provided by the Red Pitaya developers. The last version that I tested at the end of the last year was 0.96-something. The Red Pitaya application marketplace worked with that version.

For information. The 0.96 versions are still available from http://downloads.redpitaya.com/downloads

If the Red Pitaya application marketplace doesn't work anymore, then the only solution that I can propose is to create a separate micro SD card for each of my SDR projects following the corresponding 'Getting started' instructions:
http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver/#getting-started-with-sdr-and-hdsdr
http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-hpsdr/#getting-started

pavel-demin, 2017-07-12:
> Like I said, Pavel's blogs are confusing and somewhat wrong but at least it's some help...

I'd appreciate if you could point me to what is wrong in my notes and I'll try to fix it.

stevec5000, 2017-07-12:
The marketplace worked on the old 0.96 versions but then some of the apps like the logic analyzer and spectrum analyzer didn't work.
I don't understand a lot your notes but where you say to:
>Download customized SD card image zip file.
>Copy the content of the SD card image zip file to an SD card.
Saying to copy the contents of the image file to the SD card will not work.

pavel-demin, 2017-07-12:
> Saying to copy the contents of the image file to the SD card will not work.

Strange. It's exactly how it works for me. I open the .zip file with any program that can read .zip files and then copy the content of the .zip file to an empty micro SD card.

BTW. The .zip files that I provide for the SDR projects differ from the OS images currently provided by the Red Pitaya developers. The .zip files that I provide for the SDR projects contain many small files that can be copied directly to an empty micro SD card. The OS images currently provided by the Red Pitaya developers contain a single .img file that should be copied to a micro SD card with a special program.

stevec5000, 2017-07-13:
I tried that but when I remove the SD card that has the OS on it the unit will not work.

stevec5000, 2017-07-14:
I got a new SD card and copied the fines onto it but it still doesn't work, didn't think it would since you can't make it bootable by just coping some files onto it at random, but it's a useless OLD version 0.92 anyway.
I had to upgrade to the latest Beta, ver. 0.97-396 before to get the Logic Analyzer to work but now there is no access to the App Store!  I downloaded the latest Stable ver. 0.97-336 and installed it using the Win32disk Imager and now there is access to the App store but the Logic Analyzer won't run so to get the Logic Analyzer to work I have to switch back to ver. 0.97-396 again!
When are those lazy programmers going to get busy and finish this product?  I'm going to have to tell the USA distributor to stop selling them until the project is completed!

pavel-demin, 2017-07-14:
> I got a new SD card and copied the files onto it but it still doesn't work

As before, you don't provide enough details for me to understand what exactly are you doing. What is the name of the .zip file? How you check that it doesn't work? Are you following all the steps from the getting started instructions or skipping some of them?

> you can't make it bootable by just coping some files onto it at random

Actually, the boot loader that comes with the ZYNQ chips is intelligent enough to read the FAT file system and find the required `boot.bin` file. You can find the commands that Xilinx recommends to run when preparing a micro SD card at the following link:

http://www.wiki.xilinx.com/Prepare+Boot+Medium

The following commands from these instructions create a FAT formatted partition and just copy boot.bin to this partition:
```
mkfs.vfat -F 32 -n boot /dev/sdX1

mkdir -p /mnt/boot
mount /dev/sdX1 /mnt/boot

cp boot.bin /mnt/boot/
```

So, copying `boot.bin` from a .zip file to a FAT formatted micro SD card is enough to make it bootable with the ZYNQ chips.

pavel-demin, 2017-07-14:
> There are video showing it working on Youtube but they seem to all be fakes, like the ones for quadcopters that can't really fly such as the Lily drone.

> Why is a product that's been out for a couple of years still using Buggy Beta software?

> When are those lazy programmers going to get busy and finish this product? I'm going to have to tell the USA distributor to stop selling them until the project is completed!

BTW. Please refrain from posting to this issue your complaints that aren't related to my notes and projects. Reading them doesn't help me to understand your problem and to help you solving it.

closed 2017-07-19 by pavel-demin