pavel-demin/red-pitaya-notes #492: can't follow "build a bootable SD card"

laurencebarker, 2017-07-16:
I've been trying to follow the steps in "LED blinker", but I can't make the SD card. I suspect the process requires the SD card to have a certain configuration at the start, but I can't work it out. 

If I put in an SD card with two partitions on it: the script rapidly fails. It reports errors when trying to create two partitions, and the download process fails almost immediately.

If I put in an SD card with no partitions: something similar happens. 

If I have an SD card with one FAT16 partition occupying about half the card: the script runs through a process downloading lots of files taking about 30 minutes. It leaves an error at the end trying to remove the ROOT device (see these last few lines) and I still only have one partition.

What state should the card be in at the start of the process please?

> EOF_CAT
I have no name!@laurence-vm:/# 
I have no name!@laurence-vm:/# apt-get clean
I have no name!@laurence-vm:/# 
I have no name!@laurence-vm:/# echo root:changeme | chpasswd
I have no name!@laurence-vm:/# 
I have no name!@laurence-vm:/# service ntp stop
[ ok ] Stopping NTP server: ntpd.
I have no name!@laurence-vm:/# service ssh stop
[ ok ] Stopping OpenBSD Secure Shell server: sshd.
I have no name!@laurence-vm:/# 
I have no name!@laurence-vm:/# history -c
I have no name!@laurence-vm:/# 
I have no name!@laurence-vm:/# sync
I have no name!@laurence-vm:/# exit
umount: /tmp/ROOT.BwvwiETwP7: not mounted
rmdir: failed to remove '/tmp/ROOT.BwvwiETwP7': Directory not empty
zerofree: failed to open filesystem /dev/

There are two lines in the script:

boot_dev=/dev/`lsblk -lno NAME $device | sed '2!d'`
root_dev=/dev/`lsblk -lno NAME $device | sed '3!d'`

It looks like these are trying to find the second and third partitions on the memory card (but I may be wrong).

I've tried this on a native linux laptop, which does have /dev/mmcblk0, and a VM which has /dev/sdb (and its partitions become /dev/sdb1 and /dev/sdb2). Both are running ubuntu - is this part of my problem?

Laurence Barker G8NJJ

pavel-demin, 2017-07-18:
Hi Laurence,

Thanks for testing `debian.sh` with a SD card.

> I suspect the process requires the SD card to have a certain configuration at the start, but I can't work it out.

Normally, the SD card should be partitioned by the commands on lines 17-21:
https://github.com/pavel-demin/red-pitaya-notes/blob/master/scripts/debian.sh#L17-L21

So, no special configuration is required.

> It looks like these are trying to find the second and third partitions on the memory card (but I may be wrong).

The first line returned by `lsblk -lno NAME $device` is the device name. So, the two `lsblk` commands are trying to find the first and second partitions.

For example, here is the output of the `lsblk -lno NAME /dev/sda` command on one of my virtual machines:
```
$ lsblk -lno NAME /dev/sda
sda
sda1
sda2
```

> I've tried this on a native linux laptop, which does have /dev/mmcblk0, and a VM which has /dev/sdb (and its partitions become /dev/sdb1 and /dev/sdb2). Both are running ubuntu - is this part of my problem?

Normally, the script should work on both Debian and Ubuntu. I tested it on Debian 8 and Ubuntu 14.04.

I'll try to test the script later today on a VM running Ubuntu 16.04.

As a workaround, I'd suggest to try building a bootable SD card image and then write it to a SD card. That's how I'm regularly using this script.

Best regards,

Pavel

laurencebarker, 2017-07-18:
Hi Pavel

I had already tried the "make an image file" approach with scripts/image.sh. It fails with similar errors. I've done this on my VM and on a native linux laptop. The console messages are:

$ sudo sh scripts/image.sh scripts/debian.sh red-pitaya-debian-8.6-armhf.img 1024
[sudo] password for laurence: 
1024+0 records in
1024+0 records out
1073741824 bytes (1.1 GB, 1.0 GiB) copied, 5.38172 s, 200 MB/s
mkfs.fat 3.0.28 (2015-05-16)
unable to get drive geometry, using default 255/63
Auto-selecting FAT32 for large filesystem
/dev/loop0p2 has 255 heads and 63 sectors per track,
hidden sectors 0x0000;
logical sector size is 512,
using 0xf8 media descriptor, with 2064384 sectors;
drive number 0x80;
filesystem has 2 32-bit FATs and 8 sectors per cluster.
FAT size is 2013 sectors, and provides 257540 clusters.
There are 32 reserved sectors.
Volume ID is 0e2ea5a7, no volume label.
mke2fs 1.42.13 (17-May-2015)
Discarding device blocks: done                            
Creating filesystem with 12288 1k blocks and 3072 inodes
Filesystem UUID: 969eb904-1284-4134-b636-1fb65f8067d9
Superblock backups stored on blocks: 
	8193

Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (1024 blocks): done
Writing superblocks and filesystem accounting information: done

W: Cannot check Release signature; keyring file not available /usr/share/keyrings/debian-archive-keyring.gpg
I: Retrieving InRelease 
I: Failed to retrieve InRelease
I: Retrieving Release 
I: Retrieving Packages 
I: Validating Packages 
E: Couldn't download dists/jessie/main/binary-armhf/Packages
cp: cannot create regular file '/tmp/ROOT.xraa6mx62V/etc/': Not a directory
cp: cannot create regular file '/tmp/ROOT.xraa6mx62V/usr/bin/': No such file or directory
cp: cannot create regular file '/tmp/ROOT.xraa6mx62V/etc/': Not a directory
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
 94 1749k   94 1647k    0     0   558k      0  0:00:03  0:00:02  0:00:01 1662k
curl: (23) Failed writing body (767 != 16384)
chroot: failed to run command ‘/bin/bash’: No such file or directory
rm: cannot remove '/tmp/ROOT.xraa6mx62V/etc/resolv.conf': No such file or directory
rm: cannot remove '/tmp/ROOT.xraa6mx62V/usr/bin/qemu-arm-static': No such file or directory

for the memory card script:
The reason why I thought it might depend on previous memory card content is: the scrip at least runs to completion if the SD card has ONE fat16 partition on it. And the partition is still there afterwards, so the line "parted -s -s $device  mklabel msdos" does NOT erase the card, as the linux documentation suggests it should.

I'm sure this is something I'm doing wrong, but I can't find out what! I will spend this evening running the scripts line by line to see what happens. The error messages always relate to the file "/tmp/ROOT.xxxxxx"

Laurence Barker G8NJJ

dl1sdz, 2017-07-18:
Hello,

I had similar problems until I deleted all partions on the card, created
one partition and formatted this with fat. So I had to recreate the
original structure of the SD card. Afterwards I just had to follow the
scripts from Pavel.

73 de

Hajo dl1sdz

---
Cela est bien dit, mais il faut cultiver notre jardin.

http://hajos-kontrapunkte.blogspot.de/

laurencebarker, 2017-07-18:
I had tried having a single FAT partition on the card; that didn't work either.

I think I have found the problem, but I'll have to wait until later to confirm. The script creates 2 partitions:
/dev/sdb1 12MB
/dev/sdb2 7GB

the intention is that the BOOT drive should be the tiny one. But it creates them the other way round (root=ext4, 12MB; boot=FAT32, 7GB).

The reason is: on my computer the output from "lsblk" is reversed in order. The code make sense if the partitions are listed in numerical order. If I run the command "lsblk -lno NAME /dev/sdb" the output I get is:
sdb
sdb2
sdb1

and the code picks the third row as the ROOT device.

If I tell lsblk explicitly to order by the NAME field it comes out correct:

"lsblk -lno NAME -x NAME /dev/sdb" gives output
sdb
sdb1
sdb2

and I THINK (will be later tonight) the script will work ok.

So - why is my computer sorting differently? is it picking up a sort order from somewhere else?

Laurence Barker G8NJJ

pavel-demin, 2017-07-18:
I've just tried to run `sudo sh scripts/image.sh scripts/debian.sh ~/red-pitaya-debian-8.6-armhf.img 1024
` on a VM running Ubuntu 16.04. However, the `mkfs.vfat` output looks different:

```
/dev/loop0p1 has 255 heads and 63 sectors per track,
hidden sectors 0x0000;
logical sector size is 512,
using 0xf8 media descriptor, with 24576 sectors;
drive number 0x80;
filesystem has 2 16-bit FATs and 4 sectors per cluster.
FAT size is 24 sectors, and provides 6123 clusters.
There is 1 reserved sector.
Root directory contains 512 slots and uses 32 sectors.
```

and here is the corresponding lines copied from the previous comment:
```
/dev/loop0p2 has 255 heads and 63 sectors per track,
hidden sectors 0x0000;
logical sector size is 512,
using 0xf8 media descriptor, with 2064384 sectors;
drive number 0x80;
filesystem has 2 32-bit FATs and 8 sectors per cluster.
FAT size is 2013 sectors, and provides 257540 clusters.
```

Could you try to add the following commands to `debian.sh` and check their output?
```
lsblk -lno NAME $device
echo $boot_dev
echo $root_dev
```

laurencebarker, 2017-07-18:
Hi Pavel

 

Already done that… there is a more recent post from a few minutes ago

 

I was getting boot_dev was /dev/sdb2, root_dev was /dev/sdb1

(ie the wrong way round)

 

And if I ran “lsblk -lno NAME /dev/sdb” I got
sdb
sdb2
sdb1
 

I’ve added –x NAME to the lsblk command to sort it by name and it’s running now.

 

 

Also – if the partitions are already mounted, they don’t get unmounted first so they don’t get re-created. Not sure how to do that.

 

 

 

Laurence Barker

 

laurence@nicklebyhouse.co.uk

pavel-demin, 2017-07-18:
I'd say that adding `–x NAME` is the right solution for this problem. I'll update all the scripts.

pavel-demin, 2017-07-18:
`-x NAME` is added to all the scripts with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/5c0f4511a802a1b222734696441aee43a5573a55)

laurencebarker, 2017-07-19:
Pavel

thank you - that's fixed the problem. I've seen there's a lot of traffic about lsblk on the internet today - something changed last Autumn. I imagine I'll have to edit RedPitaya's scripts too.

closed 2017-07-19 by laurencebarker