pavel-demin/red-pitaya-notes #496: Diversity with Red Pitaya: Same signal in In1 and In2 ??

Jmontsecal, 2017-07-24:
Hello Pavel,
      I am using Openhpsdr mRx PS V3.42   with  Red Pitaya OS 0.97.336, connected directly via ethernet PC-Red Pitaya.  Works fine receiving, but I tried to use the  "diversity" option connecting a second antenna in the "IN2", to try noise cancelling and realized that depite selecting RX1 or RX2,  both come from the same antenna : I diconnect from IN2 and RX2 and RX1 keep on receiving the same signal.
     I have seen in some video an example where the operator was able to select  different antennad doing what I am doing, so I am a bit lost regading the use of the second antenna and diversity.
     What I do is "power", search for  a signal, activate "Diversity", select RX1+RX2 and play with the phase and amplitude . I am able to perfectly null the RX1+RX2 output, meaning to me that they are in fact the same signal. Is like if Red Pitaya were sending what samples in IN1  to IN2.
    In the configuration I selected model=HPSDR and Connection=METIS.
    So, What should be the correct steps to force Red Pitaya to use  both antennas and be able to benefit of the Diversity capabilities.
    Thanks in advance,
    Josep Maria

pavel-demin, 2017-07-24:
Could you please check what is the application that you're running on Red Pitaya?

If RX1 and RX2 have exactly the same signal, then I'd suspect that you're running 'SDR receiver compatible with HPSDR'. This application is mostly usable with the multichannel CW and RTTY skimmers.

For diversity, the correct application is 'SDR transceiver compatible with HPSDR'.

Jmontsecal, 2017-07-25:
Thanks Pavel for you hint !.
IN fact I was using the "receiver". I will try with the "transceiver".
Jma

closed 2017-07-25 by Jmontsecal