pavel-demin/red-pitaya-notes #500: question: installing development machine error

tech1962, 2017-07-26:
Hello,
I've followed the 'installing dev machine' steps and I find that when I unpack the Vivado installer (20GB download) that there is not enough room in the virtual machine storage for it to fit...according to the error messages.

Could you please suggest what steps to take to expand the storage in Virtual Box?
I am completely new to VM and so am not sure which parameters to alter so as to not affect the build.

Thanks for any help.
tech1962

pavel-demin, 2017-07-26:
Thanks for finding this problem. I agree that the recent versions of the Vivado Design Suite require more disk space and the image size of 33 GB isn't enough anymore.

I've just checked the size of `Xilinx_Vivado_SDK_2016.4_1215_1.tar.gz`, unpacked `Xilinx_Vivado_SDK_2016.4_1215_1` and `/opt/Xilinx` after the installation. The total size is 21 + 21 + 20 = 63 GB. So, looks like I should update my instructions and replace 33 GB with something like 129 GB.

I think that the easiest solution would be to recreate the virtual machine with a larger image size.

Another relatively easy solution would be to add a new virtual disk to the virtual machine and mount this disk as `/opt`. A new virtual disk can be added via Settings->Storage->Controller SATA->Second plus icon (adds hard disk).

The new virtual disk should be partitioned and formatted. It can be done from the running virtual machine with the following commands:
```
parted -s /dev/sdb mklabel msdos
parted -s /dev/sdb mkpart primary ext4 512 100%
mkfs.ext4 -F -j /dev/sdb1
```

To mount this new disk at boot the following line should be added to `etc/fstab`:
```
/dev/sdb1       /opt           ext4    defaults            0       2
```

laurencebarker, 2017-07-26:
It is possible to resize the VM images. Mine was about 60GB and I did resize it, but I was able to install Vivado before increasing its capacity. 

Tip to minimise demand while installing Vivado: 
1. Download the installer (about 20GB)
2. Unpack it to a suitable location
3. Delete the installer
4. Install Vivado
5. Delete the installation directory

To resize the Virtualbox drive: I can't tell you the exact details but it's something like this

1. Close down your VM
2. Use google to find instructions but run a windows command line program "vboxmanage modifyhd"
3. If you restart your VM you will now find that it's partition is the size it always was, but the drive it is on is larger. You need to resize the partition.

To do that you will need to do:
1. shut down your VM
2. download to your host machine a "live CD" version or "parted" or "gparted". Mine was called "gparted-live-0.28.1-1-i686.iso"
3. In the virtualbox settings, select " storage" select the optical drive, click "live CD" and click the tiny CD icon and select the iso file you have just downloaded. Check the boot order and make sure your VM is allowed to boot from a CD. 
4. restart your VM; it will now boot from the "CD" and start the partition editor. You need to select your "hard drive" partition and tell it to resize to full size. Sorry I can't tell yout he exact commands - I didn't write them down.
5. Shut down your VM, go back into virtualbox and deselect "live CD"
6. Restart your VM - it should now have access to the larger disk.

I was stumped on this for a while, because my laptop doesn't have a CD drive. But it turns out there's no actual CD involved - like the virtual machine, the CD it virtual!

tech1962, 2017-07-27:
Hello Pavel and Laurence,
Thank you both for your suggestions. I have made some good progress.
I created a new VM with 70 GB, and during the gzip and tar commands I deleted the un-needed files to save space...then deleted the old VM.
I am trying to conserve space as I'm using a 256GB Solid State drive WIN10 PC...so even 70 GB is a lot of usage.

If I may ask for some more help up this steep learning curve: I have been able to unpack the SDK to the dir as suggested but when I run the ./xsetup command I get a "no x11 display variable was set" error.

I believe this means I am not running a graphical interface for Linux ..as I'm still in the command window within the VM.

Can you advise how I get past this hurdle ? I am logged in as root user.

Thanks again.

laurencebarker, 2017-07-27:
Sorry, I've no idea what that means - I installed the full GUI version of Ubuntu. It wasn't very big. Even after the Xilinx install it was only using about 35GB.

Laurence Barker G8NJJ

pavel-demin, 2017-07-27:
> If I may ask for some more help up this steep learning curve: I have been able to unpack the SDK to the dir as suggested but when I run the ./xsetup command I get a "no x11 display variable was set" error.

Normally, I access my virtual machine via SSH with enabled X11 forwarding. To display applications with graphical user interfaces, a X11 server (Xming for MS Windows or XQuartz for Mac OS X) should be installed on the host computer.

closed 2017-08-13 by pavel-demin