pavel-demin/red-pitaya-notes #501: Red Pitaya HPSDR Board id?

laurencebarker, 2017-07-27:
Currently Pavel's code reports (using the HPSDR "discovery" reply message) that the Red Pitaya is a Metis board with attached Mercury receivers. Is there any reason why it shouldn't report that it is an Orion (which has two ADCs) or even have its own id? 

My reason for asking is: as far as I know, Power SDR doesn't care what the board id is (you have to tell it in the setup screens what the hardware is); but pihpsdr does care and determines what features to make available accordingly. So when pihpsdr adds puresignal support, that won't be available if it thinks it is talking to a Metis; and it won't forward RX attenuation values now because it thinks there is nothing attached that needs them.

As far as I can see it would make no difference at the Red Pitaya end, It wouldn't matter to Power SDR and would only matter to pihpsdr (for which I'd be happy to write the necessary code). But there could be other programs I haven't considered or other issues I haven't realised. 

I'd be happy to make an experimental build pretending to be an Orion and try it - but has this been considered before please?

Laurence Barker G8NJJ

pavel-demin, 2017-07-28:
> I'd be happy to make an experimental build pretending to be an Orion and try it - but has this been considered before please?

It would be great if you find a combination of the ID bytes that activates all the necessary functionality in pihpsdr. The current ID bytes are the first that I found to work with PowerSDR mRX PS. I never tried other combinations.

laurencebarker, 2017-07-28:
Hi Pavel

I've tried a couple of values. I've configured the board to look like both an Orion (as used in the Anan200D) and a Hermes. Both configurations work fine with both pihpsdr and PowerSDR mrx PS. In PowerSDR you still have to tell the program what the board is (in the "general" setup tab). When you tell it to use "orion" you lose the "Alex" filters page, and you can choose which receiver is connected to which ADC & the hardware doesn't support that. So that might be confusing.

I suspect the best option at the moment is to use "Hermes" with the code at line 298 being:
  uint8_t reply[11] = {0xef, 0xfe, 2, 0, 0, 0, 0, 0, 0, 32, 1};

When set to Hermes, on both programs I get RX1 coupled to ADC1 and RX2 coupled to ADC2, as you would want. The software doesn't "know" there is actually a separate ADC connected to RX2, but that doesn't seem to matter. It would only really matter if you had independent filter etc settings for both ADCs, which the protocol and PC software don't yet support.

There is no priority behind this from my perspective: I've been able to implement a change and I'm not ready to use pihpsdr fully yet anyway. (my next job will be to interface the RedPitaya to an Arduino that controls my RF front end board, at which point I will start wanting the attenuator values from pihpsdr).

Laurence Barker G8NJJ

ka6s, 2017-07-30:
Pavel,

I've done two things this weekend - Checked the change for the reply
message that was suggested - and PiHPSDR likes that very much!  I think
that is a definite suggestion for change that I would ask you to accept.

I also tried the bit file for hpsdr supplied - and I see the hermes control
codes working for filter control - but not the CW inputs. They don't seem
to be present/accounted for. I even put a printf into cw_input() routine
for if any of them being set - didn't see it.

Steve

pavel-demin, 2017-07-31:
Steve, thank you for the tests.

I've updated the board ID with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/4ce19e00633335086aed5b07dea479ce3d826e25).

I'll test the CW inputs when I'm back to my equipment.

closed 2017-08-16 by pavel-demin