pavel-demin/red-pitaya-notes #502: Unable to transmit with OpenHPSDR

Paco31, 2017-07-30:
Hi Pavel and all,
I use Red Pitaya with OpenHPSDR, and it works perfectly in reception, but I can not to transmit.
I put ExtIO_RedPitaya_TRX.dll in the directory C: \ Program Files (x86) \ OpenHPSDR \ PowerSDR mRX PS, although I think it is not necessary with this application ...
How to configure OpenHPSDR to transmit ?
Thank you for your help

pavel-demin, 2017-07-31:
> How to configure OpenHPSDR to transmit?

As far as I know, it's enough to push the MOX button in PowerSDR mRX PS to start transmitting. The TX frequency should be within the limits of one of the amateur radio bands.

Red Pitaya should be running SDR transceiver compatible with HPSDR.

> I put ExtIO_RedPitaya_TRX.dll in the directory...

ExtIO_RedPitaya_TRX.dll is for SDR transceiver but not for SDR transceiver compatible with HPSDR.

Paco31, 2017-07-31:
Thank you very much for your answer.
To check if I transmit, I listen on another receiver (I have an ANAN 100 and a Flex 1500).
 I do not hear and do not see anything, but I believe that with 10 mW this is normal .
I have connected Red Pitaya on an oscilloscope, and I have a signal ...
I will mount a small amplifier to try.
I would like to take this opportunity to thank you and congratulate you for your work
Regards

on3vna, 2017-07-31:
Paco31,

You can try with the TUNE button, then the transmitter will output a modulated signal. Eventually increase the output level of the TUNE to have more output power. (Adjust the drive level)

Rgds Johan

Paco31, 2017-07-31:
Hi Johan,
thank you for your answer.
I think the emission is correct, with the MOX or the Tune, but I thought that despite the low power (10 mW), I had to hear it on a receiver, and that is not the case.
On the oscilloscope, I get 7 V peak to peak, on a 50 ohms traversing load.
My call : F6DET
Regards

Paco31, 2017-08-02:
Hi all, 
I measured 0,7 V peak to peak, and not 7 V ...
Regards

closed 2017-08-13 by pavel-demin