pavel-demin/red-pitaya-notes #503: Question about PulsedNMR

johncprice, 2017-07-30:
Pavel, thank you very much for all of this helpful information.  I am thinking about a project similar to your PulsedNMR, but I am not sure if the Red-Pitaya hardware is suitable or not. All of my previous experience is with Altera boards. It might save me a lot of time if you could answer two questions:

1. My project requires a very stable external clock.  I see that the Red-Pitaya board has a jumper for an external clock that goes directly to the ADC.  If I supply this external clock can the same clock be used for the FPGA logic and for the DAC?  I need to have the ADC, the FPGA logic, and the DAC all synchronous.

2. In your PulsedNMR project it looks like the Tx signal is written out by the CPU.  Is the Tx signal phase coherent with the Rx local oscillator?  If not, won't the NMR data be at a different phase for every pulse?

Thanks very much for any help you are able to give.

John

pavel-demin, 2017-07-31:
Thank you for your interest in my NMR application for Red Pitaya. I started this project to help a group from an university with their pulsed NMR setup. My version of this code is still in a state of a basic prototype that lacks some important features (for example, generation of the B pulses). The missing features can be added relatively easily and I know that someone has successfully built a pulsed NMR system for student use based on my prototype.

>  If I supply this external clock can the same clock be used for the FPGA logic and for the DAC?

Yes, the external ADC clock can be used for the FPGA logic and for the DAC.

>  Is the Tx signal phase coherent with the Rx local oscillator?

I didn't know that the phase coherence was a requirement. In the current version of my code the phases of the TX and RX signals aren't synchronized.

johncprice, 2017-07-31:
Thanks very much for answering these questions. This is for a student project at the Univ. of Colorado where I teach. Red-Pitaya is the only board I have found that is inexpensive enough for such a project, so it is good news what you say about the clocks.  In simple NMR transceivers like the one we are considering the Tx oscillator is usually the same as the Rx local oscillator except maybe at a different phase. The CPU might just gate the waveform from the DDS to create the Tx pulse, rather than providing the complete RF waveform.  In more sophisticated high-field NMR systems there is usually an arbitrary waveform generator to create "shaped pulses", which is more similar to what you have done. But in those systems there is also a programmable sequencer that can trigger the RF pulses at definite times and at definite phase relative to the Rx local oscillator. I will definitely get one of these boards and try to get more familiar with the tools. Thanks again for your help!

pavel-demin, 2017-08-31:
I now have a version of the FPGA configuration with a simple pulse sequence generator. The arbitrary waveform memory is replaced with a DDS and a sequencer. Each pulse has four parameters: delay before, pulse duration, delay after, phase offset. Here is a sequence of three pulses with different phase offsets (0, pi/2, pi):
[image]

johncprice, 2017-09-03:
That looks really neat!  Are you going to post the updated code?  We have not made any progress on our project but I started playing with the RP board.

PaulMGlover, 2018-03-09:
Hi Pavel (and sorry to John for adding to a thread rather than starting my own)

I am new to RedPitaya and FPGA synthesis methods. However, I have designed NMR spectrometers over the years.  The RedPitaya and your fantastic software and instructional resource is wonderful and I am learning a lot.  I have got as far as working out how the blocks are wired together.  I can see how you have changed to a pulse generator and DDS on the tx side (John's suggestion above).  I can determine how this works and what instructions I need to feed to it to make the pulses you show above.  What I can't work out yet and can't see from your 'server' code is how the tx side FIFO works?  It appears that you have done-away with the tx-data-counter so how does the pulse generator 'know' how many words are in the FIFO (or does it matter).  Clearly the 'pulse-nmr.c' code is for the original method of arbitrary generation.  I can see how to change it but I don't know if the FIFO works in the same way as the rx side or if it is different?  

As far as I can see the FPGA is very nearly exactly what I need.  There are only one or two things I would like to implement to 'ice the cake' as it were.  One thing would be to remove the sync from every tx pulse so that the pulses are coherent thoughout the sequence (if required).  I was going to use an unused upper bit from pinc to gate it perhaps.  However, I am getting well ahead of where I am now.

If you are interested I will send you a photograph of two large circuit boards of discrete logic chips which are 30 years old and were used to implement a pulse sequence controller for an NMR imaging system!

Thank you again.

Paul

pavel-demin, 2018-03-09:
Hi Paul,

Thank you for checking the new pulse generator.

You're right. The server code isn't in sync with the FPGA configuration. I've just committed to the devel branch the code that I used to generate the three pulses shown in my previous comment. The values written to the TX FIFO buffer are on the lines 129-142:
https://github.com/pavel-demin/red-pitaya-notes/blob/develop/projects/pulsed_nmr/server/pulsed-nmr.c#L129-L142

The server code is unfinished because I don't have any idea of how to implement a user interface for the new pulse generator. Your ideas are very welcome.

The pulse generator knows the state of the TX FIFO buffer via the s_axis_tvalid wire. This wire is connected to the non-empty signal of the TX FIFO buffer. Here is a link to the code reading the data from the TX FIFO buffer:
https://github.com/pavel-demin/red-pitaya-notes/blob/develop/cores/axis_pulse_generator_v1_0/axis_pulse_generator.v#L57-L63

The sync signal can be controlled by the last unused bit from the phase offset register. This way it can be configured independently for each pulse. Actually, I don't know if the sync signal is required at all.

Best regards,

Pavel

PaulMGlover, 2018-03-12:
Dear Pavel

Thank you for the very speedy response.  I can now see exactly how the tx FIFO server side works so that is very helpful.  I was going to try and guess at something along those lines - but I have to admit that I would have got the write word order wrong (at least to start with).

Thy sync facility isn't really needed during a sequence of pulses, but only for the very first pulse if at all.  It might be desirable in some circumstances to be able to synchronise the tx and rx DDS devices if the frequencies were exactly the same.

Another point I was going to make was that the Rx acquisition block should be timed from the tx pulse generator as well - almost as if it were a pulse in itself.  This is vital for coherence of the signal so that data averaging can take place (always happens in NMR!).  I was wondering how this might be done but now I see that there is an 'enable' line originating in the server code and going to the rx fifo.  I was going to investigate the possibility of driving this from the pulse generator in the tx chain.

However, I need to get on with the day job and will have to pursue this later.  I just wanted to say thank you for responding to my question.  I certainly have some ideas about how one might create a flexible and usable interface once I understand how the FPGA code and server works.

Paul

PaulMGlover, 2018-04-04:
Pavel

Some success!  I thought you would like to see some output from the pulsed-NMR project.  The graphic shows an FID obtained after a 10 microsecond pulse at 18.45 MHz on a bench-top magnet
[image]. The pulse tip-angle is undefined for the moment.  The FID is typical of the unfiltered signal seen on the 'scope before further base-band filtering.

I have made some modifications to the gating. I have used two bits from the phase offset register to gate the RF pulse, and also gate the receive channel.  I have done the latter (for simplicity to start with) by putting the Rx-gate into the t-data-valid on the CICs (which is usually set to true permanently). This works but it would be better to gate the flow of data into the FIFO so that the FIR filter can operate all the time.  Otherwise the first points of the acquisition get mixed up with the end of the previous acquisition.  It isn't very obvious on this graphic because it is mixed up with the ring-down of the pulse etc.  The only other very minor change (just to see if I could edit and compile the FPGA) was to output the gates to LEDs.  For the moment the 'server' code just runs and dumps data to a file which I read into matlab - laborious but it made things slightly simpler for understanding what is happening.

I feel that I could now pass this on to some students as a project now I know that it does the business - thanks to your design skills.

Paul

PaulMGlover, 2018-04-04:
Sorry, forgot to say that the x-axis on the graph is milliseconds.

pavel-demin, 2018-04-05:
Thanks for sharing these results! I'm glad that the new pulser starts to be usable.

I agree that gating the RX path isn't trivial. That's why I prefer to keep it running continuously. The only application where I restart the RX modules is VNA. It's required for the phase measurements. To avoid the effects of the restarting, I just skip the first 200 microseconds. Maybe something similar could be done in the pulsed NMR application.

BTW. I didn't know that the NMR applications require the phase measurement. I thought that only the amplitude is measured.

twitzelbos, 2018-04-18:
Hello Pavel,

I also felt your example was very useful and I have been trying to reach you last November. I have students actively taking MRI images using the red pitaya, and we'll have an oral presentation on low-cost open source MRI at ISMRM in Paris this year. 

[image]
[image]
[image]

We will be making all the source available at ISMRM and since it contains your cores and still has parts of your pulsed-nmr example, I wanted to know what license you prefer and how you would like to be credited in the project.

Thomas Witzel

pavel-demin, 2018-04-18:
Hello Thomas,

Thanks for the pictures. It's great that your project is progressing well.

The license for the code running on the Red Pitaya board is MIT license:
https://github.com/pavel-demin/red-pitaya-notes/blob/master/LICENSE

Since the GUI client code is based on the Qt library it has to be under the GPLv3 license:
https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/pulsed_nmr/client/COPYING

It would be great if you could link my Red Pitaya notes (http://pavel-demin.github.io/red-pitaya-notes) from your project.

Best regards,

Pavel

twitzelbos, 2018-04-18:
Hello Pavel,

thank you for the very fast response. We will likely go with the MIT license as well for the HDL side of things. We can either fork your project, or contribute to your github. Either is fine with me in principle.

We will link to your red-pitaya-notes in all published works and websites.

Thomas

BeauWebber, 2018-06-18:
Hi Guys,
Yes what Pavel has done is great.
I have been conversing with him off-line, while I have been building up some NMR kits around the Red Petaya. I now have a receiver that I am very happy with - it has replaced my previous receivers in my lab, and is now in day-to-day use.
The transmitters are still work in progress, but I have had NMR signals from both a linear transmitter module, taking RF signals from Pavel's latest DDS transmitter, and also a digital transmitter that currently is driven by a different FPGA.
Perhaps some background would be useful - I have been building NMR spectrometers since 1972 - the first one had a 2 valve transmitter and solid-state receiver with a super-linear diode detector.
Pretty soon I had developed an all digital CMOS logic plus bipolar transistor power stage NMR transmitter (the bigger box in this picture) plus a NMR receiver with hard-ware phase-locked loop detector. 

Here is one of the last being commissioned in 1992 (about 24 were made) and I have used it for decades for my published research. Pi/2 into a 5mm water sample is less than 3 us. I am just retiring it now with the advent of these Red Petaya based spectrometers.
[image]

I have worked with Altera FPGAs for many years, and I am afraid getting to grips with the Xilinx firmware in the Red Pitaya is taking longer than I would like. 
I use the array processing language AplX to process the data, and it is also very good at creating and interacting with the GUI frontpanels. I now use it in Server-Client mode, where the Server runs in Linux in the ARM CPU, and talks via c programs with Pavel's firmware, and the Client runs on the desktop, handles the GUI and file I/O. 

Here is the new low-noise receiver, about 60 dB gain (variable), mounted on the Red Pitaya.

[image]

And here are the Receiver, Transmitter and Pulse Programmer GUI frontpanels (with fairly arbitrary settings), followed by a single-shot CP/MG echo train, using my prototype linear transmitter board. But I am very limited in how much power I can apply, the transmitter module needs a proper box to work well. Pi/2 into a 5mm water sample is currently 15 us , Pi is 30 us .

[image]

[image]

I am just starting to update my company instrumentation web page, but feel free to have a browse, there is quite a bit of older stuff on it too : http://www.lab-tools.com/instrumentation/ 

I am busy documenting things at the moment, so I hope to have more here and on the above web-site soon.
cheers,
     Dr. Beau Webber

dnhkng, 2018-08-22:
Hi NMR Experts, 

I want to build a permanent magnet NMR. Yes, its a weird hobby project, but I'm an organic chemist by training, and I now have some FPGA experience. My goal is about a 0.5 tesla device.

Working from Pavel's base, I would like to extend extend the project, and use the pynq framework to interface the PL. Basically VHDL for the fast stuff, and Python for the rest.

Can I get some input on what features to add? My current list includes:
  - slow DACs for gradient field control. (Maybe I try and build a MRI device later).
  - lock-in amplifier in the RX, like in this paper: "Simple and low-cost tabletop NMR system for chemical-shift-resolution spectra measurements", (https://doi.org/10.1016/j.jmr.2018.07.003)
  - 16 GPIOs. Because more is better.
  - LED control to monitor status.

Any suggestions would be great.
-David

BeauWebber, 2018-08-22:
Just to be quite clear, the pulsed low-field NMR spectrometers that I am developing, around the Red Pitaya, using Pavel’s firmware, is primarily a relaxation spectrometer.
I.e. adding a chemical shift resolution capability would be a second stage.
I am hoping they will be ready in the Autumn.
I am already employing the prototype of the receiver in my NMR Cryoporometer.
Any input re simple 0.5 T permanent magnets is appreciated.
Cheers, Beau

-- Kentforlife.net - the email service for alumni of the University of Kent

BeauWebber, 2018-12-09:
A pre-Christmas update on the NMR system :
Lots of progress since I last wrote.

A Red Pitaya with a ~60 dB gain pre-amp I have designed, plus Pavel Demin’s firmware, is working well,
and is now my main receiver for my NMR system in my research laboratory.

I am using the Apl language I talked about previously, but now working with Micro-Apl’s AplX :
running the Apl server in the ARM cores in the Red Pitaya, talking to C which talks to Pavel Demin’s firmware; and the Apl client on the desktop handles the GUI and the control and data handling.

I also have two NMR transmitter modules – for the high-power NMR digital transmitter, the prototype worked well, and the V4 boards are back from etching and assembly, but the machine shops are taking forever with the finishing stages of the 3D printed boxes.

I also have a lower-power NMR linear amplifier, that amplifies Ch1 output from the Red Pitaya, as generated by Pavel's code. Final boards away for etching and assembly, as are the 3D printed boxes.

I talked about these at an NMR conference in Florida, last February, and have now published a paper :

1.	Credit-card sized Field and Benchtop NMR Relaxometers using Field Programmable Gate Arrays. J. Beau W. Webber, Pavel Demin, Magnetic Resonance Imaging, 2018, DOI: 10.1016/j.mri.2018.09.018

I still have to have the EU certification done, but hope to be selling Version 1 NMR Spectrometers within about 6 months.
 
Cheers,
                Beau

twitzelbos, 2019-01-04:
Hi Beau and Pavel,

happy new year!! This is extremely exciting news. Congratulations on the publication! It's exciting to see progress. We are proceeding slowly as this a small side project for me... I'll start blogging soon on my github page, so people will see progress....Pavel, you got the new SDR redpitaya? I tried contacting redpitaya about it, but I'm not getting much information at this point...

pavel-demin, 2019-01-04:
> Pavel, you got the new SDR redpitaya?

I have a prototype of the new board. It looks exactly like the board shown on the Red Pitaya web site (https://www.redpitaya.com/n86/new-stemlab-sdr). Unfortunately, this prototype has a problem in the TX part. So, I can't test the TX functionality at the moment.

twitzelbos, 2019-01-04:
Okay, sorry to hear about that. Since I'm not doing SDR, I'm not too excited about the 122.88 MHz clock. Of course, I can make the NMR/MRI stuff work with that as well, but a 125 MHz clock would be way more convenient...

BeauWebber, 2019-01-04:
Hi Thomas,
Good to hear from you.
Linear Transmitter making progress, was getting good NMR signals with test system before Christmas. Now have the boards back and first prototype box to test it in, so hopefully will be firing it up properly over the next week or so.
Box machining my major delays at the moment.
Cheers, Beau

twitzelbos, 2019-01-04:
Hi Beau,
I would be interested in trying your hardware. As you know, I'm making all my work openly available (hardware and software), which may not entirely be aligned with your project, but I wouldn't mind having your setup in my lab. I have not had a chance to work on the amplifier yet for example. I'm using an open source 800 mW transmitter we developed based on an obsolete Motorola hybrid amp, and I'm also using a vintage ENI linear amp with a bit more oomph...

BeauWebber, 2019-01-04:
Hi Thomas,
Yes I have the receiver with low noise pre-amp, connected to the Red Pitaya RF input, that is now my main receiver in my NMR Cryoporometer. But is still waiting for a well screened box, my next design project, hopefully some time in January.
And a digital transmitter, waiting for the final machining on the box (solid box essential). Plus ideally SATA drive from the Red Pitaya, but currently drive from my own FPGA modules.
Plus the linear transmitter I just spoke about, takes drive from the Red Pitaya RF out and digital Gate (from the receiver board).
So I am hoping I will have some prototype NMR spectrometers for sale within a few months, all going well.
cheers,
                Beau

BeauWebber, 2019-02-04:
Hi Guys,
OK, My new Red Pitaya FPGA based NMR spectrometer is as of today now working well.
A good clean signal, much better signal to noise than one would expect, given that the receiver is not yet boxed. This is using the linear transmitter, with more than sufficient power for liquids work – about 25 μs for a 90° pulse.

Good clean single-shot NMR signal, noise barely visible, with 113mg water sample - sample and NMR coil much too large for this magnet, only some effectively giving a signal.

Next, complete design for the receiver box.
Then complete testing of digital transmitter, which is needed for measurements on ice and similar.
Then get certification.
Also start assembling the Cryoporometer temperature controlled probe – still awaiting a machined ceramic component.
Cheers,
            Beau

New all-in-one desktop computer, Red Pitaya, with my low-noise receiver plugged in as yet unboxed, my newly completed linear transmitter (gold-flashed box), old tuning box (zinc die-cast box) and old probe, new 24 MHz NMR magnet :
[P1060847 - CryoP-3 - Spectrometer running fine_cc_en_med.jpg]

Top graph : single shot echo.

Bottom graph : 20 averages, echo peak polynomial fitted.
[2019-02-04 - A-B-Echo 23.9106MHz NMR ss + av echo fit_sml.png]

-- Kentforlife.net - the email service for alumni of the University of Kent

BeauWebber, 2019-03-30:
Hi guys,
Now have the NMR receiver in a box (will soon be smaller), so this is the latest version of my NMR Spectrometer, based on Pavel's firmware, and with the NMR linear transmitter. Soon hopefully going for certification.
[image]

twitzelbos, 2019-05-27:
Hello guys,

Is there any news about the 16bit redpitaya? I have been trying to get info by emailing them, but can never get the real story. Do they ship? Do they work?

BeauWebber, 2019-05-27:
Not heard anything yet - Beau

BeauWebber, 2019-06-06:
Yes my 16 bit Red Pitaya came today.
One possible worry is not a problem - the physical lay-out of the E1 & E2 connectors is the same, and will fit the PCBs that I have had made to plug in.
I regret if will be a few days before I can fire it up and do some testing.

BeauWebber, 2019-06-06:
I am pleased to say my NMR Spectrometer has passed its Accreditation and EU testing with flying colours.
For minimum radiated noise it is necessary to replace the standard Red Pitaya +5V switch-mode power supply with a linear power supply. The other power supplies are already linear supplies.

pavel-demin, 2019-06-07:
For information. I've created a new GitHub repository and a new set of notes on the new STEMlab 122.88-16 SDR board. Here are the links:

https://github.com/pavel-demin/stemlab-sdr-notes
https://pavel-demin.github.io/stemlab-sdr-notes

twitzelbos, 2019-06-08:
Thanks, Pavel, much appreciated. Hopefully, I get my boards soon as well.

BeauWebber, 2019-06-15:
Hmm, no joy firing up the 16 bit Red Pitaya.
Green LED OK, No Heartbeat LED.
BT Hub does not see it.
SD card version.txt and file dates identical to an SD card from a recent 14 bit Red Pitaya.

BeauWebber, 2019-06-15:
The Lab-Tools NMR Relaxation Spectrometer is now ready.
On the left is the Red Pitaya : the Array processing AplX server runs in the Linux Arm processor; this talks to C routines, which in turn talk to Pavel's firmware. The surface mount low noise receiver, linear transmitter, and probe tuning are in the 3D printed gold flashed box.
 
[image]

[image]

[image]

[image]

Here is the Free Induction Decay of Fluorine in PTFE; This is part of an experiment to determine the crystalline / amorphous polymer ratio :

[image]

And here is a measurement of the Gaussian T2 of solid brittle ice, as a function of temperature; This gives information on the dynamics / mobility of the ice molecules :

[image]

twitzelbos, 2019-06-16:
Hello Beau,

Looks very nice, congratulations!!

Thomas

        External Email - Use Caution

[P1070068 - Assembled_br_cr_med]<https://user-images.githubusercontent.com/4470231/59528364-fc162a00-8ed5-11e9-9413-9f782748d876.jpg>

[RedPitaya_ARM-to-AplX-to-C-to-Firmware_bg]<https://user-images.githubusercontent.com/4470231/59529769-d9861000-8ed9-11e9-909c-e5dfead7b30d.png>

[RP Receiver block diagram_fixedFIR_bg]<https://user-images.githubusercontent.com/4470231/59529799-e6a2ff00-8ed9-11e9-95c1-83077d8f01e5.png>

[RP Transmitter block diagram_pipeline_v2_bg]<https://user-images.githubusercontent.com/4470231/59529822-f4f11b00-8ed9-11e9-9763-adc3670d3c7d.png>

[F - PTFE - Sealex Foam -4 - fit]<https://user-images.githubusercontent.com/4470231/59528555-89597e80-8ed6-11e9-85e4-a3b97c13749a.png>

[Solid Brittle Ice - T2 vs Temperature]<https://user-images.githubusercontent.com/4470231/59528794-22889500-8ed7-11e9-96ea-0d20b722e8a3.png>

The information in this e-mail is intended only for the person to whom it is
addressed. If you believe this e-mail was sent to you in error and the e-mail
contains patient information, please contact the Partners Compliance HelpLine at
http://www.partners.org/complianceline . If the e-mail was sent to you in error
but does not contain patient information, please contact the sender and properly
dispose of the e-mail.