pavel-demin/red-pitaya-notes #521: polarity of PTT, Dash, Dot inputs

laurencebarker, 2017-08-18:
Dear Pavel

I'm about to make a "Red Pitaya Shield" to house a CODEC module, my Arduino and the various I/O connectors for your project sdr-transceiver-hpsdr. I want to understand the polarity of the three input signals PTT in, Dash In, Dot in.

I've seen Steve KA6S "shield" schematic and his wiring suggests these inputs are active low (ie PTTin = 0v activates TX, etc). When I use test wires on my Red Pitaya, I've confirmed the same ie wire those pins to ground and the radio enters TX mode.

But from my reading of the C code, they look like they should be the opposite (for example PTTin is essentially copied to PTTout, which is definitely active high). I can't read Verilog yet, but I suspect the answer lies in gpio_debouncer.v. (Alternatively the input buffer may be inverting them??)

So that I understand, can you explain how these signals are inverted between the input pins and the C code please?

pavel-demin, 2017-08-18:
> So that I understand, can you explain how these signals are inverted between the input pins and the C code please?

They are inverted by the not_0 cell:

https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_hpsdr/block_design.tcl#L112-L117

laurencebarker, 2017-08-18:
Great - thank you. My plan is to learn the FPGA code, when I get past the other stuff! I have modified a script so I can open your projects into the Vivado interactive environment and see them in the block diagram editor - I just don't know what I'm reading yet.

In case it helps others: the conclusion here is that the hardware PTT, Dot and Dash input wires are all ACTIVE LOW, but the C code sees GPIO inputs that are active when '1'.

closed 2017-08-18 by laurencebarker

pavel-demin, 2017-08-18:
>  I have modified a script so I can open your projects into the Vivado interactive environment and see them in the block diagram editor

For information. It can be done without modifying the scripts with the following commands:
```
make NAME=sdr_transceiver_hpsdr xpr
vivado tmp/sdr_transceiver_hpsdr.xpr
```

pavel-demin, 2017-08-18:
BTW. You can find some more details about the scripts in my Club Vivado presentation (slides 90-111):
http://www.origin.xilinx.com/publications/products/vivado/club_vivado/2016/einhoven-2016-presentations.zip