pavel-demin/red-pitaya-notes #524: Suggestion: Use LED's to signalize program status and transmitting

ted051, 2017-08-21:
Hi Pavel,
it would be helpful, to signalize
- if the program (main) is running and
- if the WSPR transceiver is sending.
this could be done with the LED's at the Red Pitaya.
Best regards, Wolfgang

pavel-demin, 2017-08-21:
There was an issue with the blinking LED generating noise:
https://github.com/pavel-demin/red-pitaya-notes/issues/323

So, I've removed the blinking led code from almost all my projects.

pavel-demin, 2017-08-21:
For the WSPR transceiver configuration, I could add a LED control program similar to the GPIO control program discussed in https://github.com/pavel-demin/red-pitaya-notes/issues/314

The LED control program could be run from the scripts to switch the LEDs on and off.

pavel-demin, 2017-08-22:
Hi Wolfgang,

I see that you've posted a similar question on the Red Pitaya forum. Doesn't the solution from my last comment meet your requirements?

Best regards,

Pavel

ted051, 2017-08-22:
Hi Pavel,
I found your program gpio-output.c, but I can't find a hint, how the numbers are mapped to the internal LED's.
By the way: The used LED must not blink, only light or not.
Best regards, Wolfgang

pavel-demin, 2017-08-22:
> I found your program gpio-output.c, but I can't find a hint, how the numbers are mapped to the internal LED's.

`gpio-output.c` won't work as it is. The FPGA configuration should be modified to connect the LEDs to a register and to connect the register to the CPU bus. Then `gpio-output.c` should be renamed to something like `leds-output.c`. In `leds-output.c`, `*(uint8_t *)(cfg + 2) = (uint8_t)number;` should be replaced with something like `*(uint8_t *)(cfg + 3) = (uint8_t)number;`, where `cfg+3` is the address of the new register in the modified FPGA configuration.

I can do these modifications but I'm still waiting for you to confirm that this solution (a LED control program could be run from the scripts to switch the LEDs on and off) meets your requirements.

ted051, 2017-08-23:
Hi Pavel,
thank you for your quick reply. The intention came from Uhland, DL4UP. He and I would satisfied with your announced solution, but we also need a solution to trigger a "poweroff" or "halt" command via a switch, like I proposed in issue #380 .
Thanks in advance, Wolfgang

pavel-demin, 2017-08-23:
> we also need a solution to trigger a "poweroff" or "halt" command via a switch

Do you need this functionality for WSPR of for some other configuration(s)?

ted051, 2017-08-23:
- Only for WSPR -

waterwin, 2017-09-18:
Please do not introduce the noise led blinking in the basic programs. There are other ways to know you are transmitting.

ted051, 2017-09-18:
We don't suggest a blinking led! Only once in a minute - isn't this acceptable?

closed 2021-06-22 by pavel-demin