pavel-demin/red-pitaya-notes #532: If /bin and /sbin is not included in PATH environment variable on host system => image creation fails

RealLitb, 2017-09-03:
On Archlinux, there is no separate /bin and /sbin directory, but both are just symlinks to /usr/{bin,sbin}. So there are no /{bin,sbin} entries in PATH. But that makes image creation (I use debian.sh) fail, because the chroot won't find various basic tools like grep and cat. I suggest adding /bin and /sbin prior to doing anything in the chroot so that things work on Archlinux.

pavel-demin, 2017-09-04:
It could be great if you could contribute a fixed and tested version that works for you on Arch.

I'm not sure if I understand what exactly should be fixed. Normally, the commands on lines 65-68 in debian.sh should add /bin and /sbin to PATH:
https://github.com/pavel-demin/red-pitaya-notes/blob/master/scripts/debian.sh#L65-L68

pavel-demin, 2017-09-24:
Solved with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/24b02c75d33a64956c9849809a70c42c908b7915).

closed 2017-09-24 by pavel-demin