pavel-demin/red-pitaya-notes #534: GPS PPS signal for HPSDR transceiver

pe0mgb, 2017-09-07:
Hello Pavel,
About a year ago you told us that a GPS PPS signal (DIO3_N) for HPSDR was still on your to do list. Is this still on your to do list? 
Thank you for all the work you are doing for this beautiful project.

Best Regards,
Gert PE0MGB

pavel-demin, 2017-09-07:
Yes, it's still on my to do list.

pe0mgb, 2017-09-07:
:-) Can you tell something about the planning on this subject?

pavel-demin, 2017-09-07:
No.

pavel-demin, 2017-09-07:
Let me remind you that this is my hobby project. Working on this project is supposed to be fun and stress free. So, please, let's keep it that way. Answering questions about budgets/schedules/deadlines/etc isn't my idea of stress-free fun.

pe0mgb, 2017-09-07:
You are absolutely right with your remark. My first question was also in that way. To put any pressure at you was not my meaning and when I gave you that impression, please accept my apologies for that.
Regards,
Gert PE0MGB

f6dro, 2017-11-17:
Hello Pavel ,

I've just loaded my red Pitaya yesterday with your HPSDR compatible transceiver software , it works like a charm. Thank U ver much for your work
73 Dom/F6DRO

UNISTRAPHIOlivier, 2021-12-07:
Hello!

I just read this old post.
We use on our beacon two redpitaya with HPSDR receivers 122-16 and 125-14.
I be interested to for sync with PPS.
Thanks in advance.

73 F4HTB!

F4HTB, 2021-12-07:
Hello!

I just read this old post.
We use on our beacon two redpitaya with HPSDR receivers 122-16 and 125-14.
I be interested to for sync with PPS.
Thanks in advance.

73 F4HTB!