pavel-demin/red-pitaya-notes #545: Question: Peamp and attenuator I/O pins in E1 connector "missbehave"?

Jmontsecal, 2017-09-13:
First, I am running FW 098-336 STEM 14 as well OpenHPSDR Charly 25 rec 2017-07-20.
The I tried to use the IO-1, IO-2 and IO-3 from the Red Pitaya E1, that supposedly should control a preamplifier and a couple of attenuators, but when I try to use them  from OpenHSDR (W10), the Preampl Out (IO-1) always stays HIGH despite the level of attenuation I select, moreover, the 10dB and 20dB, IO-2 and IO-3  toggle but the sequence is not consistent with the dB level selected. 
This is the table of  combinations I get:
ATT      IO1    IO2   IO3
-36       H        H      H
-24       H        L       H
-18       H        H      H
-12       H        H      L
-6         H        L       H
 0         H         L      L
 6         H        H      L
12        H         L      H
18        H         L      L
24        H         H     L
36        H         L      L
  It happens the same with Charly25 and Hamlab/Stemlab  Hardware configurations.
 Is this a bug?, Is not implemented? Is not supposed to be used in that way?  Do I need to use a special configuration? 
Please advise,
Thanks,  Josep Maria

pavel-demin, 2017-09-13:
I'm afraid that I can't help with the versions modified by Charly25. Since I'm not the author of those modifications, I don't know what is modified and how it's supposed to behave.

I can only suggest to contact the Charly25 developers.

TheChief, 2017-09-13:
Hi Josep Maria,

please write to hamlab@redpitaya.com to get your question answered.

Best regards
Markus

Jmontsecal, 2017-09-14:
Thanks!  I will do so.
Regards,
Josep Maria

closed 2017-09-23 by pavel-demin