pavel-demin/red-pitaya-notes #554: Question about IQ transmission

Tcub, 2017-09-16:
Hello,

I have been using the Red Pitaya along with GNU Radio to generate qpsk and qam modulation successfully, but I am now looking to transmit I and Q information separately via the two transmit ports and receive separately as well to help driver another RF IC.

I tried just separating I and Q in GNU Radio and passing each to a Red Pitaya block targeting a different port but the received constellation when reconstructed is warped. I assume it is because both streams of information are not being passed to and from the Pitaya to the host machine at the same time so when reconstructing the full signal it is warped.

I was just wondering if something like this had been attempted before and if it is achievable at  the GNU Radio level, or if it would require a change to the sdr transceiver Vivado project to achieve?

pavel-demin, 2017-09-18:
I don't think that the [sdr_transceiver](http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver) projects would work for this kind of application.

However, I think that the [sdr_transceiver_wide](http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-wide) project could be easily converted into a IQ transceiver.

It should be enough to connect the currently disconnected Q streams to IN2 and OUT2.

If you're interested, I can do this modification and add [sdr_transceiver_wide](http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-wide) to the new [Alpine image](http://pavel-demin.github.io/red-pitaya-notes/alpine).

Tcub, 2017-09-18:
That would be great. I was really apprehensive about trying to rework a functioning device as I doubt I would get anywhere fast with it. 

So in that wideband project the original use is to just drive and receive from single ports. Would your modification mean that once you pass the IQ data from GNU Radio to the Pitaya, the sdr_transceiver_wide project will automatically separate IQ and then pass the Q stream to OUT2and the reverse for IN2?

pavel-demin, 2017-09-18:
After some more thought, I realized that I'm not sure if I understand how exactly the [sdr_transceiver_wide](http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-wide) project should be modified to meet your requirements.

However, I still think that this project would be the best starting point. It has already two streams (I and Q) that are processed synchronously and adding more streams shouldn't be a problem.

It'd be great if you could show me a diagram of the signal processing chain (mostly its Red Pitaya part) that you're looking for.

Tcub, 2017-09-18:
I think the image below is what I am going for. I could be wrong but I don't think the DDS and complex mixer are necessary as the chip I am interfacing with mixes to a much higher carrier anyway which was my understanding of there role in the Pitaya, to mix the IQ up to the transmit frequency set by the user in GNU Radio.

Am I making sense at all?

[image]

pavel-demin, 2017-09-19:
Great! Thanks for the image!

It indeed looks exactly like [sdr_transceiver_wide](http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-wide) with the DDS frequency set to zero and with the Q streams connected to IN2 and OUT2.

I've modified [sdr_transceiver_wide](http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-wide) and added it to the new Alpine based image:

http://pavel-demin.github.io/red-pitaya-notes/alpine

To disable DDS, the central frequency should be set to 0. To enable both I and Q streams the IQ bitmask should be set to 3.

[image]

Here is a picture of a complex 1 kHz cosine waveform as seen by an oscilloscope connected to OUT1 and OUT2:

[image]

Tcub, 2017-09-19:
That's fantastic. I will download the image and get it tested as soon as possible, Thank you very much

Tcub, 2017-09-21:
Just wanted to say thank you again. Works like a charm

closed 2017-09-21 by Tcub

sabine95, 2017-11-22:
Hey,
How can I activate the Blocks transceiver wide?