pavel-demin/red-pitaya-notes #566: question or feature request

satfan52, 2017-09-22:
I use the VNA application and a RF bridge to measure the VSWR  of my Antenna and the GAIN of some of my filters. For the VSWR, I do the usual SOL calibration with the ""DUT port" of the RF bridge and everything works well, no issues.  I want to use the same set-up to measure the Gain of different filters but I am having troubles.  Following the usual procedure, I insert the filter in serie between the output 1 port of the red-pitaya and the input of the RF bridge and I leave the "DUT port" of the RF bridge open. When I hit the Gain button I see the shape of the Gain expressed in DB (the blue curve) but this curve is not calibrated at all like it is on the standard "frequency response analyser" app for instance. For instance the "Gain" of my filter  is above 0 db on some frequencies and there is no frequency calibration at all. So the only way I can get the correct picture is by doing a scan without any Filter attached, then do another scan with the Filter and then calculate manually in Excell the GAIN of the Filter by substracting the two curves. I was wondering if there is a more efficient way than this approach and if not, if the calibration process could be improved so that the GAIN of a DUT inserted in serie shows the real performance of the DUT without any need for further manual calibration after  the measure.  I also tried with a simple T junction, so without the RF bridge and it is the same issue. Thanks

pavel-demin, 2017-09-23:
Currently, the short calibration is used for the gain calculations:
https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/vna/client/vna.py#L294

It works when used with a series fixture.

From your explanation, I understand that if a bridge is used for the gain measurements, then the open calibration should be used for the gain calculations.

I can think of two quick workarounds:
 - do short calibration with the DUT port open before the gain measurements
OR
 - edit line 294 in vna.py and replace short with open

As a solution that would work with both a bridge and a series fixture, I can add a GUI element to select the calibration that should be used for the gain calculations.

satfan52, 2017-09-23:
Thank you for the explanation. Yes the GUI update you propose is obviously the optimal approach from an end-user perspective. In the mean-time, I will just do a short calibration with the DUT port open as you suggest it.  Thanks again

pavel-demin, 2017-10-03:
I'm currently updating the VNA client code and as a solution for this issue I've added two gain plots: gain short and gain open.

Here is how the updated interface looks like:
[image]

satfan52, 2017-10-05:
Thanks, it is clear indeed. I can not test it for the time being unfortuntaly.  I will do it as soon as I can.

pavel-demin, 2017-10-05:
Thanks for confirming that it's OK. There isn't much to test at the moment. I'm still preparing a release with this new interface.

pavel-demin, 2017-10-10:
The updated VNA client is now in [this release](https://github.com/pavel-demin/red-pitaya-notes/releases/tag/20171010).

closed 2017-10-28 by pavel-demin

sdelaun, 2018-02-09:
Dear Pavel,

thanks for this great job!
I've used your last released application (Win10 Pro) & RedPitaya for the first time this week and I have an issue trying to save data in .s1p file, the window is instantly closed and the file generated file is empty (0 Ko).
I can do it in txt but the data are not formated to open it with RF soft...
Did you know this bug?
Thanks

sdelaun, 2018-02-09:
Could you present a typical application schematic with the ports used (out1, in1, etc.), best placement of T sma. I'm not familiar with VNA...
Thanks

pavel-demin, 2018-02-09:
> I have an issue trying to save data in .s1p file, the window is instantly closed and the file generated file is empty (0 Ko).

Thanks for finding this problem. I've just fixed it with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/2393bdfb5d2b7307654e0705fca3014027e1c5b5).

You can apply this fix by downloading [vna.py](https://raw.githubusercontent.com/pavel-demin/red-pitaya-notes/master/projects/vna/client/vna.py) to the VNA client directory.

> Could you present a typical application schematic with the ports used (out1, in1, etc.), best placement of T sma.

You can find some schematics in #250. More schematics can be found in the documents linked from the [VNA notes](http://pavel-demin.github.io/red-pitaya-notes/vna).

sdelaun, 2018-02-09:
Thanks, I'll try it next week!

pavel-demin, 2018-03-05:
The fixed write s1p and write s2p code is now in the [latest release](https://github.com/pavel-demin/red-pitaya-notes/releases/tag/20180305).