pavel-demin/red-pitaya-notes #584: GQRX

CyberVines, 2017-10-05:
Just curious, has GQRX support been deprecated? Also I have noticed the IMGs are now readonly and HPSDR will not open port 1001. Also I Would love to see native USB support for GQRX.

pavel-demin, 2017-10-05:
I'm not sure if I understand what is the problem. Could you please provide more details about your setup and about what exactly doesn't work? The more details the better.

Anyway, I'll try to answer what I can.

> Just curious, has GQRX support been deprecated? 

As far as I know, GQRX should still work with [SDR transceiver](http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver).

> I have noticed the IMGs are now readonly

I don't understand this part. What images? What do you mean by read-only?

> HPSDR will not open port 1001

I don't understand this part.

> Also I would love to see native USB support for GQRX.

I don't understand how it's related with my notes. None of my projects uses USB for communicating with Red Pitaya.

closed 2017-10-08 by pavel-demin

CyberVines, 2017-10-30:
Hi Pavel,

All issues have been resolved with 0.98.615 & http://bazaar.redpitaya.com

SDR transceiver compatible with HPSDR (1.0-20171025)
SDR receiver compatible with HPSDR (1.0-20171025)
SDR transceiver (1.0-20171025)

Thank you for your support!

Kind Regards,