pavel-demin/red-pitaya-notes #592: VNA build issue

G4DCP, 2017-10-09:
When I try to build the vna I  get an issue where the name vna is used as a file and a directory causing problems:- 

 ```
 adding: mcpha/index.html (deflated 60%)
  adding: mcpha/start.sh (deflated 45%)
cp: cannot overwrite non-directory 'vna' with directory 'projects/vna/bazaar'
arm-linux-gnueabihf-gcc: error: vna/src/main.c: Not a directory
arm-linux-gnueabihf-gcc: fatal error: no input files
compilation terminated.
/opt/Xilinx/SDK/2016.4/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin/../lib/gcc/arm-linux-gnueabihf/5.2.1/../../../../arm-linux-gnueabihf/bin/ld: cannot open output file vna/vna: Not a directory
collect2: error: ld returned 1 exit status
sed: can't read vna/info/info.json: Not a directory
  adding: vna (deflated 86%)
```
Have I set the build up wrongly?

Peter (G4DCP)

pavel-demin, 2017-10-09:
Hello Peter,

Thank you for finding this problem.

After rereading my building from source notes, I realized that executing the last two commands one after the other would lead to the problem that you observe.

With [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/718ee4afcbbccc8333835b55abf0d95bfcd2bea0), I've just modified `build-all.sh` and moved all the bazaar scripts to a separate script.

The good news is that the error isn't related to the building of the SD card image. So, you should already have a new SD card image called `red-pitaya-alpine-3.6-armhf-20171009.zip`.

Best regards,

Pavel

G4DCP, 2017-10-09:
Many Thanks for the quick reply

closed 2017-10-09 by pavel-demin