pavel-demin/red-pitaya-notes #594: Development Machine

jkhgroves, 2017-10-09:
Hi Pavel,

Are you going to change your development machine from Debain to Alpine as you are now running Alpine on the RP?

Kind Regards,

Julian.

pavel-demin, 2017-10-09:
Hi Julian,

I'm quite happy with my development machine running Debian. So, I don't think that I'm going to switch it to Alpine Linux.

Best regards,

Pavel

closed 2017-10-09 by pavel-demin

tknopp, 2017-11-04:
@pavel-demin: on http://pavel-demin.github.io/red-pitaya-notes/ it seems that the Debian instruction (and the image) have been removed to the alpine one. Is this intentional? I use the Debian image and am quite happy with it. Your comment above says the same :-)

pavel-demin, 2017-11-04:
This issue is about a development virtual machine running on a PC.

Please create new issues for unrelated questions.

I removed a couple of pages from my notes with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/470304b4224bdca8306dbe7f25733059f7df088a).

One of them was indeed Debian with Red Pitaya ecosystem. It was initially a prototype that I built as an example for the Red Pitaya developers. I think that these days the SD card images based on Ubuntu provided by the Red Pitaya developers should be in a good shape and there is no reason to keep my image with outdated Red Pitaya ecosystem.

There is still a SD card image based on Debian without Red Pitaya ecosystem:

http://pavel-demin.github.io/red-pitaya-notes/led-blinker/

The development machine instructions are still available:

http://pavel-demin.github.io/red-pitaya-notes/development-machine/

tknopp, 2017-11-05:
thanks! I thought this issue was about the RP image and therefore did not want to create an new issue.