pavel-demin/red-pitaya-notes #595: documentation/startup script under Alpine

F6ITU, 2017-10-09:
Hi
On three different config and 2 different places (2 different red pitaya),  a copy of the "*start.sh" file of any kind of image on the " topmost directory on the SD card" (when creating the image) doesn't start anything

could it be possible to change the "Getting start" instructions

"Optionally, to start one of the applications automatically at boot time, copy its start.sh file from apps/<application> to the topmost directory on the SD card."

into 

Optionally, to start one of the applications automatically at boot time, copy its start.sh file from apps/<application> to the /etc/local.d/ directory on the SD card.

(just a suggestion, it's probably a pebcak problem from my side, but the suggested path cannot work under Alpine Linux )

MArc

pavel-demin, 2017-10-09:
Hi Marc,

Actually, copying the start.sh file to the topmost directory on the SD card works. Here is a video showing what an where should be copied:

https://www.youtube.com/watch?v=8piEybrHvWU&t=140

The problem with `/etc/local.d/` that it doesn't exist on the SD card.

Best regards,

Pavel

pavel-demin, 2017-10-09:
BTW. There is already a script in `/etc/local.d/` that executes `start.sh` from the topmost directory on the SD card. Here is a link to the script:
https://github.com/pavel-demin/red-pitaya-notes/blob/master/alpine/etc/local.d/apps.start

So, nothing should be added to `/etc/local.d/`.

closed 2017-10-09 by pavel-demin

davidhay45, 2022-06-03:
I had a similar problem and found that the /etc/local.d/apps.start file needed to have execute permissions.  However, this problem only occurred on the 20220322 alpine sd card image file.